CadQuery/cadquery #1146: Font_FontMgr, warning: unable to find font

alinelena, 2022-09-06:
trying to use a font like this fails to load it. 

```
Fontconfig error: Cannot load default config file: No such file: (null)
Font_FontMgr, warning: unable to find font './Font_Awesome_6_Brands-Regular-400.otf' [regular]; 'DejaVu Sans' [aspects: regular,bold,italic,bold-italic] [paths: /usr/share/fonts/truetype/DejaVuSans.ttf;/usr/share/fonts/truetype/DejaVuSans-Bold.ttf;/usr/share/fonts/truetype/DejaVuSans-Oblique.ttf;/usr/share/fonts/truetype/DejaVuSans-BoldOblique.ttf] is used instead
```

is not a path or a permissions issue since i copy the font in the current place. also openscad and lualatex have no issue using it. 

[test.tar.gz](https://github.com/CadQuery/cadquery/files/9493674/test.tar.gz)

cadquery version 2.2.0b1 and fontconfig is 2.14.0

adam-urbanczyk, 2022-09-06:
You want to specify the font path (thus not name), so try using the `fontPath` parameter. 

See the docs: https://cadquery.readthedocs.io/en/latest/classreference.html#cadquery.Workplane.text

alinelena, 2022-09-07:
first fontPath containing absolute path to the font ( filename included) worked.
this is some confusing but worked.

closed 2022-09-07 by alinelena

adam-urbanczyk, 2022-09-09:
@alinelena works for me, check the preferences in CQ-editor

[image]