CadQuery/cadquery #1150: How does cut work?

xxx, 2022-09-12:
Hi,

I'm Just wanting to cut a couple of custom-shaped solids out of a larger solid, but the cut operation seems to do nothing:

```python
import cadquery as cq

length = 3 * 25.4
width = 1.375 * 25.4
height = 3 * 25.4

openingHeight = 2.6 * 25.4
endOpeningHeight = 1.4 * 25.4

wallThickness = 2

cutoutWidth = (width - (wallThickness * 3)) / 2

def makeCutout():
    nudge = 0.0
    
    cutoutVertices = [
        (length / 2 + nudge, (height / 2) - wallThickness),
        (length / 2 + nudge, ((height / 2) - wallThickness) - openingHeight),
        (-length / 2 - nudge, (-height / 2) + wallThickness),
        (-length / 2 - nudge, (-height / 2) + wallThickness + endOpeningHeight),
    ]

    return (
        cq.Workplane("XY")
        .polyline(cutoutVertices)
        .wire()
        .extrude(cutoutWidth)
        .rotate((0, 0, 0), (1, 0, 0), 90)
    )

body = cq.Workplane("XY").box(length, width, height).edges("|X and <Z").fillet(5.0)

cutout1 = makeCutout().translate((0, -width / 2 + cutoutWidth + wallThickness, 1))
cutout2 = makeCutout().translate((0, width / 2 - wallThickness, 1))

# show_object(cutout1, options={ "color": (255, 0, 0) })
# show_object(cutout2, options={ "color": (0, 0, 255) })
body = body.cut(cutout1).cut(cutout2)

show_object(body, options={ "alpha": 0.5 })
```

`nudge` is there to make the cutouts slightly larger than the base, per [this comment](https://github.com/CadQuery/cadquery/issues/529#issuecomment-738032050), but it doesn't change anything - the model is always solid when it's exported. I feel like I'm missing something small and obvious, but I'm fully clueless on what to do here.

fpq473, 2022-09-12:
Your polyline() call creates only three line segments, so the extruded solid is invalid.

One fix would be to replace the wire() call with close() -- this closes the loop, and creates a wire.

https://cadquery.readthedocs.io/en/latest/classreference.html#cadquery.Workplane.close

xxx, 2022-09-12:
@fpq473 Thanks, That does it.

I was using `wire` because I had run into [this comment](https://github.com/CadQuery/cadquery/issues/546#issuecomment-744055818) and [this comment](https://github.com/CadQuery/cadquery/issues/546#issuecomment-744548173), but I was missing the fact that those examples include the final point to overlap the initial one.

closed 2022-09-12 by xxx