CadQuery/cadquery #1152: extend tolerance or precision from sweep function

phiser678, 2022-09-13:
The sweep function is acting strange when thin! Anybody can help?

## To Reproduce

```python
```
path= cq.Workplane("XZ").spline([(0,0),(20,100/2),(0,100)])
sweep=cq.Workplane("XY").circle(1).sweep(path)

[image]
However if I use `circle(2.1)` it is correct. Seems like precision errors?

## Environment
OS: Ubuntu 22.04 
Using:
Python 3.10.4
cadquery==2.2.0b1
cq_editor/_version.py = "0.3.0dev"

lorenzncode, 2022-09-13:
This is an OCCT tessellation issue (duplicate of #969).

Tessellation impacts both STL export and CQ-editor visualization.

See the comment in #969 for STL export workaround using netgen.  Then even if you don't need STL you could visualize the model from STL file or export STEP and use another viewer.

phiser678, 2022-09-14:
Thanks for the info, so if I export it to STEP and convert it with netgen to STL it should give the correct result? Let's give that a try!

phiser678, 2022-09-14:
This code produces a valid result!
```
import netgen.occ as nocc
import netgen.meshing as meshing

s = nocc.OCCGeometry('thintube.step')
m = s.GenerateMesh(perfstepsend=meshing.MeshingStep.MESHSURFACE,maxh=.1,optsteps2d=0)
m.Export('thintube.stl', 'STL Format')
```
[image]

closed 2022-09-14 by phiser678