CadQuery/cadquery #1153: Workplane.save()

roipoussiere, 2022-09-14:
I like the simplicity of [`Assembly.save()`](https://cadquery.readthedocs.io/en/latest/classreference.html#cadquery.Assembly.save):

```py
asm = ( cq.Assembly()
    .add(box)
    .save('box.step')
)
```

To export a model, in the other hand, the syntax is not as nice:

```py
box = cq.WorkPlane().box(1, 2, 3)
cq.exporters.export(box, 'box.step')
```

I wonder if the assembly export syntax could be applied on a model as well, for consistency and simplicity:

```py
box = ( cq.WorkPlane()
    .box(1, 2, 3)
    .save('box.step')
)
```

adam-urbanczyk, 2022-11-06:
It would be nice indeed. @jmwright @lorenzncode would you agree?

lorenzncode, 2022-11-06:
Yes, that might be convenient and I think it could support saving to BREP file too!