CadQuery/cadquery #1158: Wrong face orientation when exporting assemblies to VRML

mojca, 2022-09-22:
I'm not yet "proficient in interpreting VRML files", but it seems that the assemblies exported into VRML end up with wrong face orientation.

I first reported the issue to KiCad:
* https://gitlab.com/kicad/code/kicad/-/issues/12400

and prepared the files with a minimum example here:
* https://github.com/mojca/vrml-backface-culling-issue

```python
import cadquery as cq

box = cq.Workplane().box(2.0, 4.0, 1.0)
assy = cq.Assembly(box, loc=cq.Location((0,0,0)), color=cq.Color("black"))

cq.exporters.export(box, 'box-ok.wrl', cq.exporters.ExportTypes.VRML)
assy.save('box-has-back-face-culling-issues.wrl', exportType='VRML')
```

@jmwright

adam-urbanczyk, 2022-09-23:
Currently VRML export is based on VTK and it uses VTK-based normal calculation. In 7.6 it is possible to use the normals from OCCT. This line would have to be modified.

https://github.com/CadQuery/cadquery/blob/259cc7b4e47cf8da695085ab507963b57c60e046/cadquery/occ_impl/assembly.py#L194

So long story short - let's wait till 7.6 is merged and try to fix it there.

mojca, 2022-09-26:
If I understand correctly: someone needs to finalize support for Open CASCADE version 7.6.0 in CadQuery and then you'll be able to test an alternative approach to export to VRML and see if it works better? (It's not totally unambiguous what version 7.6 refers to.) Thank you for checking this out.

adam-urbanczyk, 2022-09-26:
Indeed, [OCCT/OCP] 7.6. See: https://github.com/CadQuery/cadquery/pull/1156

adam-urbanczyk, 2022-10-04:
@mojca could you check if #1167 solves this issue?

mojca, 2022-10-04:
What is the best way to install CadQuery in a proper way, so that I will be able to test that change? (I would prefer to stick with `pip` rather than `conda`, if possible. Or I can manually edit some files if needed.)

adam-urbanczyk, 2022-10-04:
This or clone the repo and use `pip install -e .`. You might want to ignore the deps if you already installed them.

mojca, 2022-10-05:
Thank you, that seems to work fine (both the installation and rendering a complex model).

mojca, 2022-10-05:
This leaves two remaining "issues" related to KiCad:
* it would be handy to have an additional export parameter for scale or base unit `assy.save('model.wrl', exportType='VRML', scale=1/2.54)`
* KiCad developers complain about the VRML output format being heavily unoptimized, see https://gitlab.com/kicad/code/kicad/-/issues/12400

But those are pretty unrelated and I'll probably open a new issue for those.

adam-urbanczyk, 2022-10-05:
Ok, open new issues @mojca, but please use more quantitative language than "something is heavily unoptimized".

adam-urbanczyk, 2022-11-06:
Solved by #1167

closed 2022-11-06 by adam-urbanczyk