CadQuery/cadquery #1165: Cadquery fails to export correctly for multiple formats for my(admittedly weird) model, but appears ok in jupyter and when exported as stl

fozbstudios, 2022-10-03:
Hello, please let me know if you wish me to provide better formatting or more info. Before I forget, should we add .vs and .eggs to the gitignore?
For my model( a weird test to demonstrate my confusion as to why the midpoint selector is needed in the roadmap, as edges() seems to function ok in returning the midpoint, but I don't have time to write that up now), I think export behaves incorrectly. 
The model displays ok in jupyter. 

Stl exports ok I think. 
3mf and amf failed to open in prusaslicer 2.5 on windows and on cura 5.1.1
step  seems to cut the sphere at random rather than combining them(also open in mayo and freecad, same result)
not sure about the svg, being correct

[image]
[image]

Here is the code which generates the files:
```

from collections import namedtuple
from cadquery import exporters
from cadquery import Compound

Info = namedtuple("Info", "trans center text")
infos=[
    Info((2,0,0),[False, False, False],"False, False, False"),
    Info((2,0,2.5),[True, True, True],"True, True, True"),
    Info((2,0,5),[True, False, False],"True, False, False"),
    Info((2,0,7.5),[False, True, False],"False, True, False"),
    Info((2,0,10),[False, False, True],"False, False, True"),
]
results=[]
for x in infos:
    result= (
    cq.Workplane()
    .transformed((0,90,90),x.trans)
    .box(1,1,5)
    .edges("<X")
    .sphere(.3, combine=True)
    .faces(">X")
    .tag("textBase")
    .edges(">X")
    .box(.4,.4,.4, centered=x.center)
)
    result2 = result.workplaneFromTagged("textBase").center(.5,0).text(x.text,1,.3, cut=False, halign="left")
    results.append(result)
    results.append(result2)
resultShapes=map(lambda x: x.val(),results)
comp = Compound.makeCompound(resultShapes)
exporters.export(comp, 'cadqueryTest.step')
exporters.export(comp, 'cadqueryTest.stl')
exporters.export(comp, 'cadqueryTest.amf')
exporters.export(comp, 'cadqueryTest.3mf')
exporters.export(
            result,
            'cadqueryTest.svg',
            opt={
                "width": 300,
                "height": 300,
                "marginLeft": 10,
                "marginTop": 10,
                "showAxes": False,
                "projectionDir": (0.5, 0.5, 0.5),
                "strokeWidth": 0.25,
                "strokeColor": (255, 0, 0),
                "hiddenColor": (0, 0, 255),
                "showHidden": True,
            },
        )
r= show(comp) # , show_result=False)
```
I installed cadquery via conda, but since I plan on helping  with development I installed via
python version is 3.9.13
conda install --only-deps -c cadquery cadquery=master
then
 pip install --no-build-isolation --no-deps -e [my git repo local path]
I differ from 9c525bcfd7bf7f1b8b9aa4370f69cad73806f37a but it shouldn't make much difference as I only had changed the following so far
cadquery/doc/_static/cadquery_cheatsheet.html
cadquery/tests/test_selectors.py
-----
A bit off topic but if anyone has a better way of installing for development than what I did, please let me know. I am not very comfortable with conda development

fozbstudios, 2022-10-03:
I'm not sure what's with the different font

lorenzncode, 2022-10-04:
>3mf and amf failed to open 

Yes, duplicate of  issue #1162 (does not affect step, step is a separate issue)

For the step export issue, here is a minimal example:
```py
import cadquery as cq

result = cq.Workplane().box(1, 1, 5).edges(">Z").sphere(0.3, combine=True)

cq.exporters.export(result, "result.step")
```

>a better way of installing for development 

pip install in editable mode looks good

fozbstudios, 2022-10-05:
> > 3mf and amf failed to open
> 
> Yes, duplicate of issue #1162 (does not affect step, step is a separate issue)
>thanks for the info on duplicates
>
> ```python
> import cadquery as cq
> 
> result = cq.Workplane().box(1, 1, 5).edges(">Z").sphere(0.3, combine=True)
> 
> cq.exporters.export(result, "result.step")
> ```
Not sure what the minimal example has to do with step export. I know step works for other models as I have used it successfully in the past. I’m just saying for this particular model export was wrong

RagingRoosevelt, 2022-12-31:
I'm running into what might be the same issue, where exporting to `3mf` produces a corrupt file that I can't open in, for instance, 3D Builder or Cura.

My minimal example:

```python
if "show_object" not in locals():
    from cq_server.ui import show_object
import cadquery as cq

obj_1 = (
    cq.Workplane("XY")
    .box(
        length=10,width=10,height=10,
        centered=(True,True,False)
    )
)
obj_2 = (
    cq.Workplane("XY")
    .box(
        length=5,width=5,height=5,
        centered=(True,True,False)
    )
    .translate((0,0,10))
)

show_object(
    obj_1,
    name="obj_1",
    options=dict(color="gray",)
)
show_object(
    obj_2,
    name="obj_2",
    options=dict(color="black",)
)

cq.exporters.export(
    cq.Compound.makeCompound(
        map(
            lambda x: x.val(),
            [obj_1,obj_2]
        )
    ),
    'export_sample.3mf',
    tolerance=0.01,
    angularTolerance=0.1,
)
```