CadQuery/cadquery #1168: Outdated pip install command in README

fpq473, 2022-10-04:
Somewhat minor, but `pip install --pre cadquery` from the README now leads to a non-working installation.  This is because 2.2.0b1 (from August) does not have a version pin on cadquery-ocp, which is now at 7.6 (uploaded recently) instead of the required 7.5.

Options include:

- Update the README to `pip install --pre cadquery cadquery-ocp~=7.5.0`
- Release 2.2.0b2 with version pinned (e.g. fixing 2.2.0b1 in a branch, or taking 3d6d2abf16191bae729776c8523fe809e442e9f1)
- Do nothing if a release is imminent.

jmwright, 2022-10-06:
@fpq473 My preference would be the second option.

jmwright, 2022-10-10:
@fpq473 The install line `pip install --pre cadquery` from the readme should work again, and installs the alpha of cadquery-ocp 7.6.3 as well.

closed 2022-10-10 by jmwright