CadQuery/cadquery #1169: optimise files exported 2d files

roipoussiere, 2022-10-04:
Here is a basic script that produce a rounded box:

```py
import cadquery as cq
from cadquery import exporters

result = ( cq
    .Workplane('XY')
    .box(50, 50, 1)
    .edges('|Z')
    .fillet(10)
)

exporters.export(result, 'test.svg', opt={
    'width': 300,
    'height': 300,
    'marginLeft': 10,
    'marginTop': 10,
    'showAxes': False,
    'projectionDir': (0, 0, 1),
    'strokeWidth': 0.1,
    'strokeColor': (0, 0, 0),
    'hiddenColor': (0, 0, 255),
    'showHidden': False
})
```

Only one path is necessary for 2d export, but if we look at the source, we can see that many paths are created:

```xml
       <g  stroke="rgb(0,0,0)" fill="none">
			<path d="..." />
			<path d="..." />
			<path d="..." />
			<path d="..." />
			<path d="..." />
			<path d="..." />
			<path d="..." />
			<path d="..." />
       </g>
```

This can be very annoying when using a cnc or laser cutter, because the tool will make a lot of additional useless movements and wille take much more time to cut the piece.

Here is a simulation of what a machine will do on the exported svg (svg moves are replaced with lines):
[image]

adam-urbanczyk, 2022-10-05:
Use DXF for cnc work. SVG export is not meant for this, it is only OKish for visualization.

roipoussiere, 2022-10-06:
I forgot to mention that I used svg here because I can easily share the image on this issue, but the problem is the same for dxf exports.

adam-urbanczyk, 2022-10-07:
OK, I misunderstood your comment. DXF exporter translates edges to equivalent DXF entities, but the topology information is lost, because DXF does not support it. It is up to your CAM tool to correctly infer the topology and generate optimal toolpaths.

As already mentioned, SVG export is very primitive (curves are discretized to segments) and only meant for visualization.

roipoussiere, 2022-10-07:
Ok, I understand that optimizing the svg is out of the scope and I will only talk about dxf.

> It is up to your CAM tool to correctly infer the topology and generate optimal toolpaths.

This can be done in CQ.

The [exportDXF() function](https://github.com/CadQuery/cadquery/blob/master/cadquery/occ_impl/exporters/dxf.py#L123) iterates over all edges, then draw them using ezdxf.

But ezdxf doesn't optimize the paths.

For instance, this code draws 2 boxes visually similar:

```py
import ezdxf

doc = ezdxf.new()
msp = doc.modelspace()

doc.layers.add(name='bad', color=6)
doc.layers.add(name='good', color=7)

upper_left  = (0,  0)
upper_right = (50, 0)
lower_left  = (0,  50)
lower_right = (50, 50)

def move(point):
    return point[0] + 60, point[1]

msp.add_line(upper_left,  upper_right, dxfattribs={'layer': 'bad'})
msp.add_line(lower_left,  lower_right, dxfattribs={'layer': 'bad'})
msp.add_line(upper_left,  lower_left,  dxfattribs={'layer': 'bad'})
msp.add_line(upper_right, lower_right, dxfattribs={'layer': 'bad'})

msp.add_line(move(upper_left),  move(upper_right), dxfattribs={'layer': 'good'})
msp.add_line(move(upper_right), move(lower_right), dxfattribs={'layer': 'good'})
msp.add_line(move(lower_right), move(lower_left),  dxfattribs={'layer': 'good'})
msp.add_line(move(lower_left),  move(upper_left),  dxfattribs={'layer': 'good'})

doc.saveas('test.dxf')
```
[image]

But if I convert them to svg and then edit it in order display moves, we see that one is not optimized:

[image]

So my proposal is to change the order of the edges list to optimize the path.

I can give try to work on it.

adam-urbanczyk, 2022-10-07:
Does #1172 help?

> > It is up to your CAM tool to correctly infer the topology and generate optimal toolpaths.
> 
> This can be done in CQ.
> 

It *can* be done, but toolpath optimization is out of scope of CQ (CQ does not want to be a CAM tool). Better open this kind of issues with your CAM program or check out this project: https://github.com/voneiden/cq-cam .

roipoussiere, 2022-10-08:
> Does https://github.com/CadQuery/cadquery/pull/1172 help?

No (see comment on the PR).

> It can be done, but toolpath optimization is out of scope of CQ (CQ does not want to be a CAM tool). Better open this kind of issues with your CAM program or check out this project: https://github.com/voneiden/cq-cam .

I understand, but I don't work with a cnc, just a laser cutter, that doesn't require a cam tool. It can't open gcode, it's basically just a classic printer driver, with some additional features like laser speed and power. And so, as far as I understand, CAM tools can't help.

roipoussiere, 2022-10-08:
Need some obvious fixes, but I'm working on a optimizer extension for ezdxf:
[image]

jmwright, 2022-10-25:
> Use DXF for cnc work. SVG export is not meant for this, it is only OKish for visualization.

@adam-urbanczyk I just ran into a situation where the CAM software I needed to use (FlatCAM) would only accept SVG. There was no option for DXF. So some people might be limited to using SVG by their software.

adam-urbanczyk, 2022-10-25:
You could try inkscape for dxf2svg conversion. It might not be smooth, there is no direct mapping between DXF and SVG.

s-ol, 2022-11-04:
> DXF exporter translates edges to equivalent DXF entities, but the topology information is lost, because DXF does not support it.

What do you mean by "does not support it"? Currently I export DXF files, then manually open them in LibreCAD to convert the individual segments into LWPOLYLINE blocks. The [`ezdxf` library has support for this](https://ezdxf.mozman.at/docs/tutorials/lwpolyline.html), is there anything stopping us from exporting each wire as an LWPOLYLINE (as long as it contains only straight and arc segments)?

adam-urbanczyk, 2022-11-04:
> What do you mean by "does not support it"? Currently I export DXF files, then manually open them in LibreCAD to convert the individual segments into LWPOLYLINE blocks and adjust the units. Then i save the file as a DXF again - so those changes could (should?) be made by the exporter.

AFAICT there is no generic wire container for all entities.

> The [`ezdxf` library has support for this](https://ezdxf.mozman.at/docs/tutorials/lwpolyline.html), is there anything stopping us from exporting each wire as an LWPOLYLINE (as long as it contains only straight and arc segments)?

"as long as it contains only straight and arc segments" is the issue.

> BTW I can +1 the need for robust SVG output. I work in a FabLab and we exclusively use SVG for Lasercutting, it's the most compatible format and the easiest to edit, and it is perfectly capable of storing the toolpaths and topology.

I looked around and did not find a nice library for dxf2svg. Maybe something you could contribute as a standalone library so that CQ and other projects could benefit? BTW: SVG does not support NURBS, so it is not the most compatible in terms of supported geometry.

If you want to contribute a better OCCT to SVG converter, that would be also possible. You'd likely have to use the some approximations (I think OCCT does have functions for that: `ShapeCustom_BSplineRestriction`).

lmarval, 2022-11-15:
Hi, 
I am am a new to the world of CAD/CAM tools and libraries, but allow me to ask,
Why optimization of svg is outside scope? 

The current implementation yields very heavy  `.svg` files which are difficult to process.
My guess is that few optimization can make a huge difference in the file size. 
If my guess is correct, why such optimization would be out of consideration?