CadQuery/cadquery #1171: dxf export: support layers

roipoussiere, 2022-10-07:
ezdxf [supports dxf layers](https://ezdxf.readthedocs.io/en/stable/concepts/layers.html).

For instance, this code generates a dxf file with 256 layers, one for each available color:

```py
import ezdxf
import random

doc = ezdxf.new()
msp = doc.modelspace()

for color in range(256):
    layer_name = f'layer{ color }'
    doc.layers.add(name=layer_name, color=color)

    start = random.randint(0, 100), random.randint(0, 100)
    end = random.randint(0, 100), random.randint(0, 100)
    msp.add_line(start, end, dxfattribs={'layer': layer_name})

doc.saveas('test.dxf')
```

Result (after importing the dxf with Inkscape):
[image]

Multi-layers dxf can be used with cnc or laser cutters in order to specify which tool or options must be used for each (ex: one layer to engrave, another one to cut).

Each layers supports:
- a color: which is a number between 0 and 255, corresponding to an index in the [color palette](https://ezdxf.readthedocs.io/en/stable/concepts/aci.html#aci);
- a [line type](https://ezdxf.readthedocs.io/en/stable/concepts/linetypes.html) (continuous, dashed, etc.);
- a [line weight](https://ezdxf.readthedocs.io/en/stable/concepts/lineweights.html), which is a value between 0 and 211 where 1mm = 100 units.

Those options could be passed in the `opt` export attribute:

```py
cq.exporters.export(shape, 'test.dxf', opt={
    'color': 5,
    'line_type': 'dashed',
    'line_weight': 25
})
```

For multi-layers export, I'm not sure what is the best syntax. I have two approaches in mind:

1. pass lists for each option:

```py
cq.exporters.export([shape1, shape2, shape3], 'test.dxf', opt={
    'color': [5, 6, 7],
    'line_type': ['continuous', 'dot', 'dashed'],
    'line_weight': [18, 25, 40]
})
```

2. Add a new class dedicated to DXF export:

```py
layer1 = DxfLayer(shape1, 5, DxfLayer.LineType.CONTINUOUS, 18)
layer2 = DxfLayer(shape2, 6, DxfLayer.LineType.DOT, 25)
layer3 = DxfLayer(shape3, 7, DxfLayer.LineType.DASHED, 40)
cq.exporters.export([ layer1, layer2, layer3 ], 'test.dxf')
```

I would like to work on this, since I need this for my project, so any advice on the export syntax will be welcome.

adam-urbanczyk, 2022-11-06:
Maybe something like (1), but pass a `Dict[cq.Shape, Dict[str,Any]]` that maps shapes to their option dicts?

sethfischer, 2023-02-10:
@roipoussiere I've been working on a utility class for multilayer DXF documents. See  #1267. I've taken the approach not the change the behaviour of `cadquery.exporters.export` but `cadquery.exporters.export` has been refactored to call the utility class.

PR #1267 (2023-03-16): DXF multilayer support
A proposed approach to #1171.

Also supports:

* units (see #1188, #1259)
* setting color by layer
* setting line type by layer

Closes #1171
Closes #1188
Closes #1259

## Todo

* [X] documentation
* [X] tests

## Notes

* The [ezdxf](https://ezdxf.readthedocs.io/) document object is available via the public API so most features of ezdxf are accessible.
* `cadquery.occ_impl.exporters.dxf.exportDXF` has be refactored but there is **no change** to the public API.
* Methods `_dxf_*` have been refactored to better facilitate unit testing. There is **no change** to algorithm  [... cut]
closed 2023-03-16 by adam-urbanczyk