CadQuery/cadquery #1174: ImportError, cannot import name 'BRepAdaptor_HCurve'

billtubbs, 2022-10-09:
## To Reproduce

I installed CADquery for the first time using

```console
pip install --pre cadquery
```

```console
pip list

Package                       Version
----------------------------- ---------
cadquery                      2.2.0b0
cadquery-ocp                  7.6.3a0
```

The error occurs when I do the import in Python script or notebook
```python
import cadquery as cq
```

```lang-none
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/Users/billtubbs/opt/anaconda3/envs/occ/lib/python3.9/site-packages/cadquery/__init__.py", line 14, in <module>
    from .occ_impl.shapes import (
  File "/Users/billtubbs/opt/anaconda3/envs/occ/lib/python3.9/site-packages/cadquery/occ_impl/shapes.py", line 59, in <module>
    from OCP.BRepAdaptor import (
ImportError: cannot import name 'BRepAdaptor_HCurve' from 'OCP.BRepAdaptor' (unknown location)
```

## Environment

OS:
- Mac OS X 10.15.7
- Python version 3.9.13

bragostin, 2022-10-09:
Same error here

adam-urbanczyk, 2022-10-09:
@jmwright looks like an issue with the pip package?

bragostin, 2022-10-09:
In my case I want to install `conda install statmodels` and I get
```
The following NEW packages will be INSTALLED:

  patsy              pkgs/main/linux-64::patsy-0.5.2-py39h06a4308_1
  statsmodels        pkgs/main/linux-64::statsmodels-0.13.0-py39h7f8727e_0

The following packages will be UPDATED:

  certifi                          2022.9.14-py39h06a4308_0 --> 2022.9.24-py39h06a4308_0
  occt                                     7.5.3-hc019471_1 --> 7.6.2-h665cfa0_3
  ocp                conda-forge::ocp-7.5.3.0-py39h1c2f70e~ --> cadquery::ocp-7.6.2.alpha-py39_0
  proj                     pkgs/main::proj-7.2.0-h1217e81_1 --> conda-forge::proj-8.2.0-h277dcde_0
  vtk                             9.0.1-qt_py39hd1b08ba_210 --> 9.1.0-no_osmesa_py39h62d5dbf_100
```
and then I get the `BRepAdaptor_HCurve` error.

jmwright, 2022-10-09:
Probably related to #1168 

I'll hopefully have time to fix this tomorrow.

adam-urbanczyk, 2022-10-09:
> In my case I want to install `conda install statmodels` and I get
>
> ```
> The following NEW packages will be INSTALLED:
>
>   patsy              pkgs/main/linux-64::patsy-0.5.2-py39h06a4308_1 [...]

If you are using conda, then reinstall cadquery after clearing package cache: https://github.com/CadQuery/cadquery/wiki/Conda-tricks

bragostin, 2022-10-09:
@adam-urbanczyk reinstall cadquery after clearing package cache worked, thank you!

billtubbs, 2022-10-09:
> @billtubbs install via `conda` or with `cadquery-ocp==7.5`

I uninstalled with `pip uninstall cadquery` and then reinstalled using:
```console
conda install -c cadquery -c conda-forge cadquery=master
```
as explained in the [documentation](https://cadquery.readthedocs.io/en/latest/installation.html#command-line-installation).

This seemed to work although I got some unusual warnings this time when running `import cadquery`:
```lang-none
objc[62378]: Class vtkCocoaTimer is implemented in both /Users/<myusername>/opt/anaconda3/envs/occ/lib/libvtkRenderingUI-9.1.9.1.0.dylib (0x1155ac208) and /Users/<myusername>/opt/anaconda3/envs/occ/lib/python3.9/site-packages/vtkmodules/.dylibs/libvtkRenderingUI-9.1.9.1.0.dylib (0x17a62e208). One of the two will be used. Which one is undefined.
objc[62378]: Class vtkCocoaFullScreenWindow is implemented in both /Users/<myusername>/opt/anaconda3/envs/occ/lib/libvtkRenderingOpenGL2-9.1.9.1.0.dylib (0x114ef4618) and /Users/<myusername>/opt/anaconda3/envs/occ/lib/python3.9/site-packages/vtkmodules/.dylibs/libvtkRenderingOpenGL2-9.1.9.1.0.dylib (0x17a582548). One of the two will be used. Which one is undefined.
objc[62378]: Class vtkCocoaServer is implemented in both /Users/<myusername>/opt/anaconda3/envs/occ/lib/libvtkRenderingOpenGL2-9.1.9.1.0.dylib (0x114ef4640) and /Users/<myusername>/opt/anaconda3/envs/occ/lib/python3.9/site-packages/vtkmodules/.dylibs/libvtkRenderingOpenGL2-9.1.9.1.0.dylib (0x17a582570). One of the two will be used. Which one is undefined.
objc[62378]: Class vtkCocoaGLView is implemented in both /Users/<myusername>/opt/anaconda3/envs/occ/lib/libvtkRenderingOpenGL2-9.1.9.1.0.dylib (0x114ef4690) and /Users/<myusername>/opt/anaconda3/envs/occ/lib/python3.9/site-packages/vtkmodules/.dylibs/libvtkRenderingOpenGL2-9.1.9.1.0.dylib (0x17a5825c0). One of the two will be used. Which one is undefined.
```

Anyhow, the previous ImportError does not occur and cadquery seems to be working. Thanks.

billtubbs, 2022-10-09:
Update: I started fresh by recreating a clean environment and installing cadquery and I don't get those package warnings as in above comment.

billtubbs, 2022-10-09:
B.t.w. the only reason I installed initially using pip is that this is the first installation method listed in the ReadMe on the GitHub site.  The conda install option is listed below that and so I missed it.

jmwright, 2022-10-10:
The install line `pip install --pre cadquery` from the readme should work again.

I don't have a preference on whether the conda or pip install methods are at the top of the readme, but we've found from experience that there is no order that will please everyone.

closed 2022-10-10 by jmwright

adam-urbanczyk, 2022-10-10:
TBH `conda` is better supported and IMO it should be described first.

jmwright, 2022-10-10:
If someone wants to submit a PR to switch the order, or to work it into another PR, I'm not opposed to it.

billtubbs, 2022-10-10:
I don't think the order is important but often pip is listed first.  Maybe add the line "CadQuery can be installed using pip or conda" at the top of the installation section so readers at least realize the first option is not the only option.

billtubbs, 2022-10-10:
Looks like my above comment has already been addressed! On the GitHub ReadMe at least.

This documentation page also has installation instructions which only mention conda at the moment:
- https://cadquery.readthedocs.io/en/latest/installation.html

Also, note that the following text appears in the bug report template:
```
Was CadQuery installed using Conda?: <!-- Note Conda is the only official installation method. If you are not using Conda, please provide details of your build/install/packaging methods, since they are often relevant to bugs and crashes. -->
```