CadQuery/cadquery #1175: 3D viewer image does not fill frame after resizing window

billtubbs, 2022-10-09:
## To Reproduce

I am launching CQ-editor from the console
```console
CQ-editor
```

When I resize the window (or change the proportions of the internal frames within the window) the rendered 3D image does not adjust to the new frame size.

[image]

If I quit QT-editor, reopen and then render it fills the window.

## Environment

OS X 10.15.7
```
conda install -c cadquery -c conda-forge cq-edimaster
```

billtubbs, 2022-10-09:
B.t.w. the only reason I installed initially using pip is that this is the first installation method listed in the ReadMe on the GitHub site.  The conda install option is listed below that and so I missed it.

jmwright, 2022-10-10:
I think this would be better filed in the CQ-editor issue tracker. This repo is just for the core library.

https://github.com/CadQuery/CQ-editor

closed 2022-10-10 by billtubbs