CadQuery/cadquery #1180: Sketch polygon requires a closed list of points

lorenzncode, 2022-10-15:
A question on Discord came up regarding `Sketch.polygon` and calling with a list of points that were not closed.

It wasn't clear to the user that there is no option to close and they are responsible for providing a closed list of points.

```py
pts = [
    (0, 0),
    (5, 0),
    (5, 5),
    (0, 5),
    (0, 0),  # user is responsible for closing
]

s = cq.Sketch().polygon(pts)

r = cq.Workplane().placeSketch(s).extrude(1)
```

I've added enhancement and docs labels.  This is either an enhancement (1) or resolved with doc update (2).

1.  At the OCCT level, `BRepBuilderAPI_MakePolygon` provides an option to close and `Close()` method.  
    Potentially this option could be exposed in CQ in both `Wire.makePolygon` and `Sketch.polygon`
2. or a note might be added to the docs that the user is responsible for providing a closed list of points.

jmwright, 2022-10-27:
I like option 1.

adam-urbanczyk, 2022-10-27:
+1 for 1

PR #1257 (2023-02-06): Add close option to Wire.makePolygon
Close wire when creating polygonal face in Sketch

resolves #1180
closed 2023-02-06 by jmwright