CadQuery/cadquery #1182: ‘pip install --pre cadquery’ appears to be installing version 1

jchidley, 2022-10-18:
```bash
pip install --pre cadquery --no-cache-dir
Collecting cadquery
  Downloading cadquery-2.2.0b2-py3-none-any.whl (142 kB)
     ━━━━━━━━━━━━━━━━━━━━ 142.2/142.2 kB 5.0 MB/s eta 0:00:00
Collecting path
  Downloading path-16.5.0-py3-none-any.whl (26 kB)
Collecting ezdxf
  Downloading ezdxf-1.0.0b0-cp310-cp310-musllinux_1_1_x86_64.whl (3.6 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━ 3.6/3.6 MB 24.1 MB/s eta 0:00:00
Collecting nptyping==2.0.1
  Downloading nptyping-2.0.1-py3-none-any.whl (19 kB)
Collecting nlopt
  Downloading nlopt-2.6.2.tar.gz (2.0 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━ 2.0/2.0 MB 29.3 MB/s eta 0:00:00
  Preparing metadata (setup.py) ... done
Collecting cadquery
  Downloading cadquery-2.2.0b1-py3-none-any.whl (140 kB)
     ━━━━━━━━━━━━━━━━━━ 140.0/140.0 kB 153.5 MB/s eta 0:00:00
  Downloading cadquery-2.2.0b0-py3-none-any.whl (137 kB)
     ━━━━━━━━━━━━━━━━━━ 137.3/137.3 kB 125.5 MB/s eta 0:00:00
Collecting multimethod
  Downloading multimethod-1.9-py3-none-any.whl (10 kB)
Collecting cadquery
  Downloading cadquery-1.2.5-py3-none-any.whl (103 kB)
     ━━━━━━━━━━━━━━━━━━ 103.1/103.1 kB 136.9 MB/s eta 0:00:00
Requirement already satisfied: pyparsing in ./cq/lib/python3.10/site-packages (from cadquery) (3.0.9)
Installing collected packages: cadquery
Successfully installed cadquery-1.2.5
```

```python
import cadquery
WARNING: CadQuery 1.x is no longer supported. Please upgrade to CadQuery 2.x https://github.com/CadQuery/cadquery
Traceback (most recent call last):
  File "/home/jack/cq/lib/python3.10/site-packages/cadquery/freecad_impl/__init__.py", line 152, in <module>
    import FreeCAD
ModuleNotFoundError: No module named 'FreeCAD'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/jack/cq/lib/python3.10/site-packages/cadquery/__init__.py", line 6, in <module>
    from .freecad_impl.geom import Plane,BoundBox,Vector,Matrix,sortWiresByBuildOrder
  File "/home/jack/cq/lib/python3.10/site-packages/cadquery/freecad_impl/__init__.py", line 154, in <module>
    path = _fc_path()
  File "/home/jack/cq/lib/python3.10/site-packages/cadquery/freecad_impl/__init__.py", line 146, in _fc_path
    raise ImportError('cadquery was unable to determine freecad library path')
ImportError: cadquery was unable to determine freecad library path
```

jchidley, 2022-10-19:
@fpq473 The machine above is an Intel i3 based x64 Alpine Linux distribution.

jmwright, 2022-10-19:
@jchidley Does the following work for you?
```
pip install --pre git+https://github.com/CadQuery
```

jchidley, 2022-10-19:
```bash
pip install --pre git+https://github.com/CadQuery
Collecting git+https://github.com/CadQuery
  Cloning https://github.com/CadQuery to /tmp/pip-req-build-u1_6imb2
  Running command git clone --filter=blob:none --quiet https://github.com/CadQuery /tmp/pip-req-build-u1_6imb2
  remote: Not Found
  fatal: repository 'https://github.com/CadQuery/' not found
  error: subprocess-exited-with-error
  
  × git clone --filter=blob:none --quiet https://github.com/CadQuery /tmp/pip-req-build-u1_6imb2 did not run successfully.
  │ exit code: 128
  ╰─> See above for output.
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: subprocess-exited-with-error

× git clone --filter=blob:none --quiet https://github.com/CadQuery /tmp/pip-req-build-u1_6imb2 did not run successfully.
│ exit code: 128
╰─> See above for output.

note: This error originates from a subprocess, and is likely not a problem with pip.
```

Or did you mean:

```bash
pip install --pre git+https://github.com/CadQuery/cadquery.git
Collecting git+https://github.com/CadQuery/cadquery.git
  Cloning https://github.com/CadQuery/cadquery.git to /tmp/pip-req-build-53qae5id
  Running command git clone --filter=blob:none --quiet https://github.com/CadQuery/cadquery.git /tmp/pip-req-build-53qae5
  Resolved https://github.com/CadQuery/cadquery.git to commit 8ddad3b71c108cd8a73cd1a7fb2c57349d5083
  Preparing metadata (setup.py) ... done
ERROR: Could not find a version that satisfies the requirement cadquery-ocp<7.7,>=7.6 (from cadquery) (from versions: non
ERROR: No matching distribution found for cadquery-ocp<7.7,>=7.6
```

jmwright, 2022-10-19:
@jchidley Are you running the following line before trying the install?

```
python3 -m pip install --upgrade pip
```

If not, try running that first. I find that I don't have problems with an updated pip version.

jchidley, 2022-10-19:
```bash
python3 -m pip install --upgrade pip
Requirement already satisfied: pip in ./cq/lib/python3.10/site-packages (22.3)
```

jmwright, 2022-10-20:
@jchidley Interesting. I'll need to set up Alpine in a VM to see if I can replicate.

jmwright, 2022-10-20:
I was able to reproduce the problem of pip defaulting to CQ 1.x on an Alpine VM install. It seems to be because pip can't install other packages for CQ 2.x (casadi, cadquery-ocp), and so it falls back to 1.x.

[image]

When I try to force it to install cadquery 2.x, I get an error saying there is no matching distribution found for casadi.
```
python3 -m pip install cadquery=2.2.0b2
ERROR: No matching distribution found for casadi
```

If I try to install casadi manually, I get the same error.
```
pythhon3 -m pip install casadi
```
Sometimes pip will try to install cadquery-ocp before casadi, and then I get a `no matching distribution found` error for cadquery-ocp instead.

@fpq473 Do you have any ideas? I've tried every variation of the pip install that I can think of, including forcing the install repo to `https://pypi.org/simple`, but nothing has worked. Could it be an incompatibility with the `manylinux` tag that is on cadquery-ocp and casadi?

Here's the `uname -a` from my VM, and I don't think it shows anything out of the ordinary.

[image]

jchidley, 2022-10-20:
Every Linux distribution is unique. One of Alpine’s quirks is that it doesn’t use `glibc` or `systemd`. It is, however, widely used for Docker images.

fpq473, 2022-10-20:
Looks like Alpine requires musllinux wheels (output of `pip debug --verbose`) and cadquery-ocp does not provide this.  Neither does casadi, nlopt, or numpy. So probably the easiest solution here is to use conda.

adam-urbanczyk, 2022-10-20:
I'm not sure that the conda packages support musl either; they are definitely not buil0/tested against musl.

nomad-vagabond, 2022-10-21:
no issues on Manjaro
`Successfully installed cadquery-2.2.0b2 cadquery-ocp-7.6.3a0 casadi-3.5.6rc2 ezdxf-1.0.0b0 multimethod-1.9 nlopt-2.7.1 nptyping-2.0.1 path-16.5.0 typing-extensions-4.4.0 typish-1.9.3
`

jchidley, 2022-10-23:
I tried various ways to get this to work on `Alpine` (there are compatibility layers for `musl`/`glibc`) without success. In the end, I installed `Arch Linux` in a `chroot` under `Alpine` and that does work: `Arch Linux` is `glibc` based.

I am closing this issue.

closed 2022-10-23 by jchidley

PWrzesinski, 2022-11-15:
Casadi doesn't seem to support Python 3.10: https://github.com/casadi/casadi/issues/2877 - this might be the reason for your issue