CadQuery/cadquery #1183: Cutting a sphere by a box fails in one direction

bernhard-42, 2022-10-27:
**Test code**

```python
import cadquery as cq

show_object(
    cq.Workplane().sphere(1).cut(
        cq.Workplane().box(2, 2, 0.5)
    ).translate((0,0,2))
)
show_object(
    cq.Workplane().sphere(1).cut(
        cq.Workplane().box(0.5, 2, 2)
    )
)
```

**Result:**
The top cut is ok, the bottom cut is broken

[image]

**Environment:**

CadQuery:
```bash
Successfully installed cadquery-2.2.dev531+g03c3266
```

adam-urbanczyk, 2022-10-27:
Try with `clean=False` as a workaround.

bernhard-42, 2022-11-05:
Sorry for the late reply, missed your answer.

Do you mean
```python
import cadquery as cq

obj = cq.Workplane().sphere(1).cut(
    cq.Workplane().box(0.5, 2, 2),
    clean=True
)
show_object(obj)
```

It still shows the wrong behaviour

adam-urbanczyk, 2022-11-05:
I meant `clean=False` (see above).

bernhard-42, 2022-11-05:
Sorry, `clean=False` works, my bad

bernhard-42, 2022-11-05:
Please feel free to close the issue, if you don't want to track the workaround.

Thank you!

adam-urbanczyk, 2022-11-05:
The issue is fine, those things do get improved with newer OCCT releases.