CadQuery/cadquery #1187: Modelling bug?

giannissc, 2022-11-04:
import cadquery as cq
from cadquery.selectors import *

from dataclasses import dataclass

# Parameters
    
@dataclass
class StatorParameters:
    diameter: float = 51.0
    hole: float = 12.0
    slot_diameter: float = 2.0
    slot_height: float = 5.0
    slot_thick: float = 5.0
    port_thick: float =  2.0
    cup_thick: float = 2.0
    wall_distance: float = 1.0
    slot_num: int = 16
    
@dataclass
class MoldParameters:
    gap: float = 10.0
    wall_thick = 3.0

mold = MoldParameters()
stator= StatorParameters()

height = stator.slot_thick + stator.port_thick + stator.cup_thick

part = cq.Workplane('XY').polarArray((stator.diameter/2 - stator.slot_height/2 - stator.wall_distance), 0, 360, 16).slot2D(stator.slot_height, stator.slot_diameter).extrude(stator.slot_thick)

foo = part.faces('>Z').workplane().polarArray((stator.diameter/2- stator.slot_diameter/2 - stator.wall_distance), 0, 360, 16).circle(stator.slot_diameter/2).extrude(stator.port_thick)

giannissc, 2022-11-04:
I have the following code but it is throwing a `Standard_NullObject: BRep_Tool:: TopoDS_Vertex hasn't gp_Pnt`

However as soon as I move the cylinders on the last line to any other radius location it can render the shape

I believe this might be a bug in cadquery as I have created the same object in Freecad without issue

giannissc, 2022-11-04:
Aligning the cylinders to the both edges of the slot seems to be causing issues

```
import cadquery as cq
from cadquery.selectors import *

from dataclasses import dataclass

# Parameters
    
@dataclass
class StatorParameters:
    diameter: float = 51.0
    hole: float = 12.0
    slot_diameter: float = 2.0
    slot_height: float = 5.0
    slot_thick: float = 5.0
    port_thick: float =  2.0
    cup_thick: float = 2.0
    wall_distance: float = 1.0
    slot_num: int = 16
    
@dataclass
class MoldParameters:
    gap: float = 10.0
    wall_thick = 3.0

mold = MoldParameters()
stator= StatorParameters()

height = stator.slot_thick + stator.port_thick + stator.cup_thick

part = cq.Workplane('XY').polarArray((stator.diameter/2 - stator.slot_height/2 - stator.wall_distance), 0, 360, 16).slot2D(stator.slot_height, stator.slot_diameter).extrude(stator.slot_thick)

part = part.faces('>Z').workplane().polarArray((stator.diameter/2- 2*stator.slot_diameter - stator.wall_distance), 0, 360, 16).circle(stator.slot_diameter/2).extrude(stator.port_thick)
``` 
[image]

adam-urbanczyk, 2022-11-06:
Tyr with `clean=False` in the last line:

```python
part = part.faces('>Z').workplane().polarArray((stator.diameter/2- 2*stator.slot_diameter - stator.wall_distance), 0, 360, 16).circle(stator.slot_diameter/2).extrude(stator.port_thick, clean=False)
```

giannissc, 2022-11-09:
I have already been using the clean=False flag but shouldn't it be able to clean the surfaces?

adam-urbanczyk, 2022-11-09:
Yes.

lorenzncode, 2023-07-17:
It looks like there is improvement with OCP 7.7.1 #1343.

[image]