CadQuery/cadquery #1189: Import DXF file failed

pengchenghu428, 2022-11-05:
ValueError                                Traceback (most recent call last)
/var/folders/6s/th2rrqjd41zgbwqzjbcb6rc00000gp/T/ipykernel_51844/3931043985.py in <cell line: 1>()
----> 1 part = cq.importers.importDXF(p_dir, tol=1)
      2 type(part)

~/opt/anaconda3/envs/py38/lib/python3.8/site-packages/cadquery/occ_impl/importers/__init__.py in importDXF(filename, tol, exclude, include)
     81     """
     82 
---> 83     faces = _importDXF(filename, tol, exclude, include)
     84 
     85     return cq.Workplane("XY").newObject(faces)

~/opt/anaconda3/envs/py38/lib/python3.8/site-packages/cadquery/occ_impl/importers/dxf.py in _importDXF(filename, tol, exclude, include)
    194         if name.lower() in selected:
    195             res = _dxf_convert(layers[name], tol)
--> 196             wire_sets = sortWiresByBuildOrder(res)
    197             for wire_set in wire_sets:
    198                 faces.append(Face.makeFromWires(wire_set[0], wire_set[1:]))

~/opt/anaconda3/envs/py38/lib/python3.8/site-packages/cadquery/occ_impl/shapes.py in sortWiresByBuildOrder(wireList)
   3601 
   3602     # make a Face, NB: this might return a compound of faces
-> 3603     faces = Face.makeFromWires(wireList[0], wireList[1:])
   3604 
   3605     rv = []

~/opt/anaconda3/envs/py38/lib/python3.8/site-packages/cadquery/occ_impl/shapes.py in makeFromWires(cls, outerWire, innerWires)
   2385 
   2386         if innerWires and not outerWire.IsClosed():
-> 2387             raise ValueError("Cannot build face(s): outer wire is not closed")
   2388 
   2389         # check if wires are coplanar

ValueError: Cannot build face(s): outer wire is not closed

adam-urbanczyk, 2022-11-05:
Looks like your drawing has wires that are not closed. Can you share it?

pengchenghu428, 2022-11-06:
Yes, it turns out that one of the widgets on the diagram didn't close

closed 2022-11-06 by pengchenghu428