CadQuery/cadquery #1197: cutting operation TopoDS_Vertex hasn't gp_Pnt

mukral, 2022-11-20:
Hello, 
I'm new to cadquery and can not get a cutting with a circle profile working (test1). A "cylindric" cut out works fine (test2, test1 also works in Freecad). What I'm doing wrong? Error is TopoDS_Vertex hasn't gp_Pnt, but, sorry, I don't understand what it means and why is it different to the cylindric cut out...

outerring = cq.Workplane('XY').circle(20).circle(30).extrude(10).translate((0,0,-5))
cuttingring1 = (cq.Workplane('XZ').center(18,0).circle(3).sweep(cq.Workplane('XY').circle(16), isFrenet=True))
test1=outerring.cut(cuttingring1) #not working 
cuttingring2 = cq.Workplane('XY').circle(18).circle(25).extrude(6).translate((0,0,-2))
#test2=outerring.cut(cuttingring2) #working

Thank you, Markus

lorenzncode, 2022-11-21:
`cut` by default calls `Shape.clean()`.  This is an OCCT level operation and in some cases it fails.

Try with clean=False:

```test1 = outerring.cut(cuttingring1, clean=False)```

lorenzncode, 2023-07-16:
I can reproduce this with OCCT/OCP 7.6.3.  I cannot reproduce with cadquery version 2.2 or 2.3 (OCP 7.7.0).