CadQuery/cadquery #1199: use constraints to place a cut

ndaman, 2022-11-24:
Is it possible to use constraints to place objects for a cut operation? Or else some kind of undocumented way to "subtract" instead of "add" in an Assembly?

Jojain, 2022-11-24:
No, the concept of an assembly is to position finished parts, so if you need to remove material from a shape you should do it before the assembly step.

ndaman, 2022-11-24:
Is there a way to use constraints outside of the assembly? They are often
more convenient and reliable than translate and rotate operations.

adam-urbanczyk, 2022-11-25:
So as @Jojain mentioned this is not currently possible, but it crossed my mind to eventually implement some editing operations.

You could extract the locations from the assy object (`cq.Assembly.loc`), apply the on the individual components and then cut.

ndaman, 2022-11-25:
Cool, thanks!