CadQuery/cadquery #1200: Trying to understand how to control the concavity of a `radiusArc`

RagingRoosevelt, 2022-11-27:
If I draw a `radiusArc` with 

```python
if "show_object" not in locals():
    from cq_server.ui import ui, show_object
import cadquery as cq

obj = (
    cq.Workplane("XY")
    .radiusArc((3,3),3)
    .close()
    .extrude(2)
)

show_object(obj)
```

I get the following:
[image]

How would I get the arc to curve the other way?  I think I could probably use `threePointArc` to accomplish this with a extra calculations.  But I'd like to figure out how to do this with the `radiusArc` method as it seems like it should support it and it would make it a lot easier to continue the following lines after the arc.  I haven't been able to find mention of a flag in the docs that would toggle the side the arc traces on.

Goal:
[image]

adam-urbanczyk, 2022-11-27:
It is in the docs...
https://github.com/CadQuery/cadquery/blob/03c3266b8d61d487349a15f73d3b3cb13bce625f/cadquery/cq.py#L2164

RagingRoosevelt, 2022-11-28:
Thanks!  I'd been reading the [readthedocs page](https://cadquery.readthedocs.io/en/latest/classreference.html#cadquery.Workplane.radiusArc) and got tunnel vision looking for a parameter for it.  I failed to read the additional notes.

closed 2022-11-28 by RagingRoosevelt