CadQuery/cadquery #1206: Assembly and part names does not appear after step file import

tommy3001, 2022-12-17:
When I generate an assembly in cadquery and export it as step, the assembly and part names does not appear in Freecad. 
Example Code:

  import cadquery as cq
  from cadquery import exporters
 #jupyter IDE
  from jupyter_cadquery import (
      open_viewer, show, web_color,
      set_defaults, get_defaults, 
      PartGroup, Part, Faces, Edges
  )
  cone = cq.Solid.makeCone(1, 0, 2) 
  assy = cq.Assembly(name="myCone")
  assy.add(
      cone,
      loc=cq.Location(cq.Vector(0, 0, 0), cq.Vector(1, 0, 0), 180),
      name="cone0",
      color=cq.Color("green")
  )
  assy.add(cone, name="cone1", color=cq.Color("blue"))
  
  assy.save('assy.step')

Result after import in Freecad:

[image]

lorenzncode, 2022-12-17:
What settings are defined in FreeCAD Import preferences?

Open Preferences to view the STEP Import settings:
Edit->Preferences...
Select Import-Export
Select STEP tab

With the default  STEP import settings I have: 
[image]

tommy3001, 2022-12-17:
Got it.  

You have to load dependent workbenches additionally before access to some extended preferences. Here, the step import preferences are only visible when part workbenches are loaded before. After loading, everything was working fine. 

[image]

closed 2022-12-17 by tommy3001