CadQuery/cadquery #1209: Selecting and tagging a face from within those returned by workplane.faces("%CYLINDER")

alikureishy, 2022-12-22:
Here's my code:
```
cylinder = (
    cq.Workplane()
    .circle(20)
    .circle(5)
    .extrude(20)
    
    .faces("%CYLINDER")       # <------ Want to select and then .tag("inner").end()
)
show(cylinder)
```

This code selects **two** cylindrical faces. I want to tag the inner face with "inner" and the outer face with "outer". I couldn't find a working selector that would select one or the other of the %CYLINDER faces for this. I tried adding selection of an n-th face using various versions of ```.faces(">Z[0]")```, but all of those calls return an error : ```ValueError: Can not return the Nth element of an empty list```.

How should I model the above tag assignments?

alikureishy, 2022-12-22:
To elaborate:
- "inner" --> tag for the cylindrical face with the smaller radius
- "outer" --> tag for the cylindrical face with the larger radius

I found a ```RadiusNthSelector``` selector class, but that only applies to Wires or Edges, as per the documentation. Can't find any other class that might work with cylindrical faces.

Furthermore, would the solution code be the same for tagging an n-th concentric conical surface?

lorenzncode, 2022-12-23:
Here is an example using `AreaNthSelector`.:

```py
import cadquery as cq
from cadquery.selectors import AreaNthSelector

cylinder = cq.Workplane().circle(20).circle(5).extrude(20)

inner = cylinder.faces("%CYLINDER").faces(AreaNthSelector(0))
outer = cylinder.faces("%CYLINDER").faces(AreaNthSelector(1))

# for CQ-Editor:
show_object(cylinder, name="cylinder")
show_object(inner, name="inner")
show_object(outer, name="outer")
```

With tags:
```py
import cadquery as cq
from cadquery.selectors import AreaNthSelector

cylinder = (
    cq.Workplane()
    .circle(20)
    .circle(5)
    .extrude(20)
    .tag("cyl")
    .faces("%CYLINDER", tag="cyl")
    .faces(AreaNthSelector(0))
    .tag("inner")
    .faces("%CYLINDER", tag="cyl")
    .faces(AreaNthSelector(1))
    .tag("outer")
    .solids(tag="cyl")   # or omit or .end(<N>)
)

# for CQ-Editor:
show_object(cylinder, name="cylinder")   # or select from tag cyl
show_object(cylinder.faces(tag="inner"), name="inner")
show_object(cylinder.faces(tag="outer"), name="outer")
```

alikureishy, 2022-12-23:
Many thanks @lorenzncode ! That worked!

closed 2022-12-23 by alikureishy