CadQuery/cadquery #1212: What is the difference between "Fixed" and "FixedPoint" constraints?

alikureishy, 2022-12-24:
[The Assembly documentation is excellent. Kudos to the cadquery team! :)]

In the code examples under the Assembly [documentation](https://cadquery.readthedocs.io/en/latest/assy.html#fixedpoint), there are "Fixed" and "FixedPoint" constraints being used. For example:

```
import cadquery as cq

b1 = cq.Workplane().box(1,1,1)
b2 = cq.Workplane().rect(0.1, 0.1).extrude(1,taper=-15)

assy = (
    cq.Assembly()
    .add(b1,name='b1')
    .add(b2, loc=cq.Location(cq.Vector(0,0,4)), name='b2', color=cq.Color('red'))
)

# fix the position of b1
assy.constrain('b1','Fixed')                      <------------- "Fixed" used here, but not documented anywhere
# fix b2 bottom face position (but not rotation)
assy.constrain('b2@faces@<Z','FixedPoint',(0,0,0.5))   <------------- "FixedPoint" used here, as per documentation
# fix b2 rotational degrees of freedom too
assy.constrain('b2','FixedRotation',(45,0,45))

assy.solve()
```

However, the [documentation](https://cadquery.readthedocs.io/en/latest/assy.html#fixedpoint) does not describe the "Fixed" constraint.
- Am I looking in the wrong place?
- What is the difference between "Fixed" and "FixedPoint"?
- Could the documentation be updated with a section for the "Fixed" constraint?

Thanks in advance.

adam-urbanczyk, 2022-12-25:
`Fixed` fixes all 6 degrees of freedom and `FixedPoint` fixes only 3 translational degrees of freedom.

alikureishy, 2022-12-25:
Thanks @adam-urbanczyk

closed 2022-12-25 by alikureishy