CadQuery/cadquery #1214: Mating surfaces around a central point/axis

alikureishy, 2022-12-27:
Here's a ```parts()``` function that produces 3 objects: ```base```, ```left_cradle``` and ```right_cradle```. Assume that I don't have access to this function's implementation (it is encapsulated), and must work with only the returned workplane objects.

```
def parts() -> Tuple[cq.Workplane, cq.Workplane, cq.Workplane]:
  cradle = (
      cq.Workplane()
      .circle(radius=outer_diameter/2)
      .circle(radius=inner_diameter/2)
      .extrude(until=cradle_length)
  )
  split_cradle = (
      cradle
      .workplane()
      .transformed(
          rotate=(0, 90, 0),
          offset=(0, 0, 0),
      )
      .split(keepTop=True, keepBottom=True)
  )
  
  # 3 produced parts:
  right_cradle = split_cradle.item(0)
  left_cradle = split_cradle.item(1)
  base = (
      cq.Workplane()
      .circle(radius=outer_diameter/2)
      .extrude(until=base_length)
  )
  return (base, left_cradle, right_cradle)
```

Invoking these methods returns the following 3 objects:

**Base**
[image]

**Left Cradle**
[image]

**Right Cradle**
[image]

Using the ```Assembly``` class, I want to mate the above objects together to produce something like below:
[image]

**I attempted this using the following code:**
```
base, left_cradle, right_cradle = parts()

assembly = (
    cq.Assembly()
    .add(left_cradle, name="left_cradle")
    .add(right_cradle, name="right_cradle")
    .add(base, name="base")
    
    # Constraints
    .constrain(
        "left_cradle", 
        left_cradle.faces(">X").faces(">Y").val(), 
        "right_cradle", 
        right_cradle.faces(">X").faces(">Y").val(),
        "Axis",
        param=pi
    )
    .constrain(
        "left_cradle", 
        left_cradle.faces(">X").faces("<Y").val(), 
        "right_cradle", 
        right_cradle.faces(">X").faces("<Y").val(),
        "Axis",
        param=pi
    )
    .constrain(
        "base", 
        base.faces(">Z").val(), 
        "right_cradle", 
        right_cradle.faces("<Z").val(),
        "Axis",
        param=pi
    )
    .constrain(
        "base", 
        base.faces(">Z").val(), 
        "left_cradle", 
        left_cradle.faces("<Z").val(),
        "Axis",
        param=pi
    )
    .solve()
)
```

... but I get this:
[image]

I known I need a couple more constraints, but I can't figure out what they should be. Since I don't have access to any information within the ```parts()``` method, I don't have the measurements to use as offsets. All I have are the workplanes, and I'd like to use the simplest approach possible to put this together.

Any suggestions? Thanks in advance.

lorenzncode, 2023-01-02:
Here is one approach you might try:

```py
assembly = (
    cq.Assembly()
    .add(left_cradle, name="left_cradle")
    .add(right_cradle, name="right_cradle")
    .add(base, name="base")
)

assembly.constrain("left_cradle", "Fixed")

assembly.constrain(
    "right_cradle@faces@>Z",
    "FixedAxis",
    (0, 0, 1),
)

assembly.constrain(
    "left_cradle@faces@>XY",
    "right_cradle@faces@<X and >Y",
    "Plane",
)

assembly.constrain("base@faces@>Z", "left_cradle@faces@<Z", "Axis")

v0 = left_cradle.vertices(">X and <YZ")
v1 = left_cradle.vertices(">XY and <Z")
e0 = cq.Edge.makeLine(v0.val().toTuple(), v1.val().toTuple())
assembly.constrain(
    "base",
    base.faces(">Z").val(),
    "left_cradle",
    e0,
    "Point",
)

assembly.solve()

show_object(assembly)
```

alikureishy, 2023-01-06:
Thanks @lorenzncode ! That worked!

closed 2023-01-06 by alikureishy