CadQuery/cadquery #1221: Example for revolve provides 3-tuples but links to revolve doc saying 2-tuples for axis of rotation

jabowery, 2023-01-11:
An example for revolve provides 3-tuples:
[image]
but [the document says to use 2-tuples](https://cadquery.readthedocs.io/en/latest/classreference.html?highlight=revolve#cadquery.Workplane.revolve):

```
axisStart (tuple, a two tuple) – the start point of the axis of rotation

axisEnd (tuple, a two tuple) – the end point of the axis of rotation
```

lorenzncode, 2023-01-11:
Also see #1125.  In this case I would probably remove the type field in the docstring.

lorenzncode, 2023-02-07:
The docstrings/docs have been updated.  The type hints are displayed in both the signature and the parameter list.

closed 2023-02-07 by lorenzncode