CadQuery/cadquery #1222: Format documentation code with blacken-docs

sethfischer, 2023-01-11:
[Blacken-docs](https://github.com/adamchainz/blacken-docs) will run Black on Python code blocks in documentation files.

This looks like a reliable way to format documentation code. Blacken-docs is used by readthedocs/readthedocs.org on their own documentation.

## Caveats

I have only identified one caveat:

1. blacken-docs [will not format default language code-blocks](https://github.com/adamchainz/blacken-docs/issues/195) `::`. For example:

```rst
The most common use is::

    s = Workplane("XY")
```

So, in order to run blacken-docs over the `cadquery/` directory (without manual intervention) instances of the above will need to be changed to:

```rst
The most common use is:

.. code-block:: python

    s = Workplane("XY")
```

sethfischer, 2023-07-09:
#1220 formats code blocks defined with `.. code-block:: python`

Do we want to change default code blocks `::` to `.. code-block:: python` so they can be formatted with [Blacken-docs](https://github.com/adamchainz/blacken-docs)?

jmwright, 2023-07-09:
After reading the comments in the discussion you linked, I would be fine with changing it. The new format is more verbose, but also more explicit.

adam-urbanczyk, 2023-07-16:
According to the README `::` is supported
```
 If your project uses Sphinx and such a configuration, add --rst-literal-blocks to also format such blocks.
```

sethfischer, 2023-07-16:
@adam-urbanczyk That's great news. Support for `::` as added to blacken-docs 2023-01-16, five days I opened this issue.

I'll mark #1220 as draft while I revert `::` -> `.. code-block:: python`  changes.

closed 2023-07-22 by jmwright