CadQuery/cadquery #1224: Sketch.hull() causes a ZeroDivisionError instead of some guidance around what specifically went wrong

alikureishy, 2023-01-12:
## To Reproduce

I was going through the Sketch tutorial, and playing around with hull(), and encountered this error. What am I doing wrong in this script below?

```python
cast: cq.Sketch = (
    cq.Sketch()
    .trapezoid(4, 3, 90)
    .vertices()
    .circle(.5, mode='s')
    .vertices()
    .circle(0.4, mode='a')
    .reset()
    .edges()
    .hull()          <----------- ZeroDivisionError
)
```

This is the model right *before* the ```hull()``` invocation above:
[image]

## Backtrace

```
ZeroDivisionError                         Traceback (most recent call last)
/var/folders/fs/jcstsp1j5fn8rz8bhfqxsglh0000gn/T/ipykernel_33196/1777645838.py in <cell line: 1>()
      8     .reset()
      9     .edges()
---> 10     .hull()
     11 )
     12 show(cast)

~/anaconda3/envs/cad/lib/python3.10/site-packages/cadquery/sketch.py in hull(self, mode, tag)
    502 
    503         if self._selection:
--> 504             rv = find_hull(el for el in self._selection if isinstance(el, Edge))
    505         elif self._faces:
    506             rv = find_hull(el for el in self._faces.Edges())

~/anaconda3/envs/cad/lib/python3.10/site-packages/cadquery/hull.py in find_hull(edges)
    393 
    394         for e in entities:
--> 395             angle, segment = get_angle(current_e, e)
    396             angles.append(angle if angle >= current_angle else inf)
    397             segments.append(segment)

~/anaconda3/envs/cad/lib/python3.10/site-packages/cadquery/hull.py in get_angle(current, e)
    310             return arc_pt(current, e)
    311         else:
--> 312             return arc_arc(current, e)
    313 
    314 

~/anaconda3/envs/cad/lib/python3.10/site-packages/cadquery/hull.py in arc_arc(a1, a2)
    245     elif r1 < r2:
    246         arc_tmp = Arc(a2.c, r2 - r1, a2.a1, a2.a2)
--> 247         xtmp1, ytmp1, xtmp2, ytmp2 = _pt_arc(a1.c, arc_tmp)
    248 
    249         delta_r = r2 - r1

~/anaconda3/envs/cad/lib/python3.10/site-packages/cadquery/hull.py in _pt_arc(p, a)
    189     l = sqrt(dx ** 2 + dy ** 2)
    190 
--> 191     x1 = r ** 2 / l ** 2 * dx - r / l ** 2 * sqrt(l ** 2 - r ** 2) * dy + xc
    192     y1 = r ** 2 / l ** 2 * dy + r / l ** 2 * sqrt(l ** 2 - r ** 2) * dx + yc
    193     x2 = r ** 2 / l ** 2 * dx + r / l ** 2 * sqrt(l ** 2 - r ** 2) * dy + xc

ZeroDivisionError: float division by zero
```

[... 364 lines cut ...]
vtk                       9.1.0           qt_py310h414f686_204    conda-forge
watchdog                  2.1.9           py310h5027ba8_0    conda-forge
wcwidth                   0.2.5              pyhd3eb1b0_0  
webcolors                 1.12                     pypi_0    pypi
webencodings              0.5.1           py310hecd8cb5_1  
websocket-client          0.58.0          py310hecd8cb5_4  
websockets                10.3                     pypi_0    pypi
werkzeug                  2.2.2                    pypi_0    pypi
whatthepatch              1.0.2              pyhd8ed1ab_0    conda-forge
wheel                     0.37.1             pyhd3eb1b0_0  
widgetsnbextension        3.6.1                    pypi_0    pypi
wrapt                     1.14.1          py310h6c45266_0    conda-forge
wurlitzer                 3.0.2              pyhd8ed1ab_2    conda-forge
x264                      1!161.3030           h0d85af4_1    conda-forge
xacro                     1.13.3                   pypi_0    pypi
xz                        5.2.5                hca72f7f_1  
yaml                      0.2.5                h0d85af4_2    conda-forge
yapf                      0.32.0             pyhd8ed1ab_0    conda-forge
zeromq                    4.3.4                h23ab428_0  
zipp                      3.9.0              pyhd8ed1ab_0    conda-forge
zlib                      1.2.12               h4dc903c_2  
zstd                      1.5.2                hcb37349_0
```

Using: jupyter-cadquery on Jupyter Notebook

crides, 2023-04-29:
not exactly the same problem, but I had trouble using `hull` today with this example:
```py
out = cq.Sketch().circle(rod_d / 2 + wall_t).push([mount_loc]).circle(mount_d / 2 + wall_t).edges().hull()
```
which is basically a big and small circle that intersect each other. using hull produced this error:
```
Traceback (most recent call last):
  File "/home/haoqing/phone-mount.py", line 10, in <module>
    sk = cq.Sketch().circle(rod_d / 2 + wall_t).push([mount_loc]).circle(mount_d / 2 + wall_t).edges().hull().reset().circle(rod_d / 2, "s").push([mount_loc]).circle(mount_d / 2, "s")
  File "/home/haoqing/.local/lib/python3.10/site-packages/cadquery/sketch.py", line 508, in hull
    rv = find_hull(el for el in self._faces.Edges())
  File "/home/haoqing/.local/lib/python3.10/site-packages/cadquery/hull.py", line 395, in find_hull
    angle, segment = get_angle(current_e, e)
  File "/home/haoqing/.local/lib/python3.10/site-packages/cadquery/hull.py", line 312, in get_angle
    return arc_arc(current, e)
  File "/home/haoqing/.local/lib/python3.10/site-packages/cadquery/hull.py", line 262, in arc_arc
    dx /= l
ZeroDivisionError: float division by zero
```
where `l` is basically the distance between the objects.

After messing around a bit, I solved _my_ issue (without understanding the code too much, just messing and observe what works) with [this commit](https://github.com/crides/cadquery/commit/bf4dde2ed6c8c99ebf4a2b70334dda29ce815d0e). _Possibly_ this could help you.

adam-urbanczyk, 2023-04-30:
AFAICT the issue occurs only with intersecting circles. Do you confirm?

crides, 2023-04-30:
I think the issue is with concentric arcs (haven't tested yet, I could do that tomorrow), and intersecting circles would be one instance of that

crides, 2023-05-05:
I tried to come up with other scenarios, but they don't seem to hit the same code path (haven't dug deeper to see what's actually going on).

Two things I tried:
```python
cq.Sketch().circle(5).push([(5, 0)]).rect(10, 10, mode="i").reset().circle(2).hull() # concentric big semi + small circle
cq.Sketch().arc((0, 0), 5, 0, 180).arc((0, 0), 5, 180, 360).reset().edges().hull()) # 2 halves of a circle
```

GabrielBlackfox, 2024-07-02:
Hi.

I've ran into problems with hull() when working on a model. I did a bit of testing with intersecting circles. The samples and code are here:

`
    r = 11.0
    d = 30.0

    sketch = cq.Workplane('XY').sketch()

    # Base circles
    sketch = sketch.push(((0, 0),)).circle(r)
    sketch = sketch.push(((0, d),)).circle(r)
    sketch = sketch.push(((d, d),)).circle(r)
    sketch = sketch.push(((d, 0),)).circle(r)

    # # Choose one of the following shapes placed between the circles to get different results

    # # Large circle, cuts deeply into base circles
    # # Result: ZeroDivisionError
    # sketch = sketch.push(((d/2, d/2),)).circle(2*r)

    # # Medium circle, tangent to base circles
    # # Result: ZeroDivisionError
    # sketch = sketch.push(((d/2, d/2),)).circle(d/sqrt(2)-r)

    # # Small circle, does not connect with base circles
    # # Result: Computes correctly
    # sketch = sketch.push(((d/2, d/2),)).circle(r/2)

    # # Large square, cuts deeply into base circles
    # # Result: ZeroDivisionError
    # sketch = sketch.push(((d/2, d/2),)).rect(2*r, 2*r)

    # # Medium square, single pint touching each base circle
    # # Result: ZeroDivisionError
    # sketch = sketch.push(((d/2, d/2),)).rect((d-sqrt(2)*r), (d-sqrt(2)*r))

    # # Small square, does not connect with base circles, but wider than the gap between them
    # # Result: Loops infinitely with full load
    # sketch = sketch.push(((d/2, d/2),)).rect(d-2*r+1, d-2*r+1)

    # # Very small square, edges do not coincide with base circles
    # # Result: Computes correctly
    # sketch = sketch.push(((d/2, d/2),)).rect(r/2, r/2)

    sketch = sketch.select().hull()

    solid = sketch.finalize().extrude(5)

    solid.val().exportStl('./test.stl')
`

There are 4 equidistant circles  (on a square). I place a single shape in the middle. Whenever the center circle is intersecting or is tangent to the others, the code fails. However, a new issue shows up when using a rectangle in the center. If the rectangle is not intersecting, but has sides bigger than gap between the circles (d-2*r) the code falls into an infinite loop. This behavior is not present when placing a similar sized circle.

Maybe this is some new info to work with