CadQuery/cadquery #1228: Cannot create dev environment

YHRen, 2023-01-13:
following the instruction `conda env create -n cq-dev -f environment.yml` and encountered this error:

```
Pip subprocess error:
ERROR: Ignored the following versions that require a different python version: 0.7 Requires-Python ~=2.7, >=3.4
ERROR: Could not find a version that satisfies the requirement cadquery-ocp<7.7,>=7.6 (from cadquery) (from versions: 7.5.3, 7.6.3a0)
ERROR: No matching distribution found for cadquery-ocp<7.7,>=7.6

failed

CondaEnvException: Pip failed
```

adam-urbanczyk, 2023-01-13:
Which conda version are you using? `pip` reqs of cq should be ignored:

https://github.com/CadQuery/cadquery/blob/369fcd95280631b1ae61ef6502ee49216ecd5cd4/environment.yml#L28

YHRen, 2023-01-14:
@adam-urbanczyk 

```
ᐅ conda -V
conda 22.11.1
```

```
ᐅ conda info

     active environment : base
            shell level : 1
          conda version : 22.11.1
    conda-build version : 3.22.0
         python version : 3.9.13.final.0
       virtual packages : __archspec=1=x86_64
                          __cuda=11.4=0
                          __glibc=2.31=0
                          __linux=5.4.0=0
                          __unix=0=0
  conda av metadata url : None
           channel URLs : https://repo.anaconda.com/pkgs/main/linux-64
                          https://repo.anaconda.com/pkgs/main/noarch
                          https://repo.anaconda.com/pkgs/r/linux-64
                          https://repo.anaconda.com/pkgs/r/noarch
               platform : linux-64
             user-agent : conda/22.11.1 requests/2.28.1 CPython/3.9.13 Linux/5.4.0-137-generic ubuntu/20.04.5 glibc/2.31
                UID:GID : 1000:1000
             netrc file : None
           offline mode : False
```

after commented out line 28, the error is the same. 

```
ᐅ conda env create -n cq-dev -f environment.yml
Collecting package metadata (repodata.json): done
Solving environment: done

Downloading and Extracting Packages

Preparing transaction: done
Verifying transaction: done
Executing transaction: done
Installing pip dependencies: \ Ran pip subprocess with arguments:
['/home/yren/anaconda3/envs/cq-dev/bin/python', '-m', 'pip', 'install', '-U', '-r', '/home/yren/github/cadquery/condaenv.ilo1_ug2.requirements.txt', '--exists-action=b']
Pip subprocess output:
Obtaining file:///home/yren/github/cadquery (from -r /home/yren/github/cadquery/condaenv.ilo1_ug2.requirements.txt (line 1))
  Preparing metadata (setup.py): started
  Preparing metadata (setup.py): finished with status 'done'
Collecting sphinxcadquery
  Using cached sphinxcadquery-1.3.3.tar.gz (264 kB)
  Preparing metadata (setup.py): started
  Preparing metadata (setup.py): finished with status 'done'
Collecting multimethod<2.0,>=1.7
  Using cached multimethod-1.9.1-py3-none-any.whl (10 kB)

Pip subprocess error:
ERROR: Ignored the following versions that require a different python version: 0.7 Requires-Python ~=2.7, >=3.4
ERROR: Could not find a version that satisfies the requirement cadquery-ocp<7.7,>=7.6 (from cadquery) (from versions: 7.5.3, 7.6.3a0)
ERROR: No matching distribution found for cadquery-ocp<7.7,>=7.6

failed

CondaEnvException: Pip failed
```

I don't see where pip can find `cadquery-ocp==7.7.x`, the latest one in the registry is 7.6.3. 
https://pypi.org/project/cadquery-ocp/7.6.3a0/

Thank you.

YHRen, 2023-01-14:
@adam-urbanczyk 
I also gave it try on my m2 mac, it has the same issue. 
```
~/github/CadQuery ᐅ conda info

     active environment : None
            shell level : 0
       user config file : /Users/yren/.condarc
 populated config files : /Users/yren/.condarc
          conda version : 22.11.1
    conda-build version : 3.21.4
         python version : 3.8.13.final.0
       virtual packages : __archspec=1=x86_64
                          __osx=10.16=0
                          __unix=0=0
       base environment : /usr/local/anaconda3  (writable)
      conda av data dir : /usr/local/anaconda3/etc/conda
  conda av metadata url : None
           channel URLs : https://repo.anaconda.com/pkgs/main/osx-64
                          https://repo.anaconda.com/pkgs/main/noarch
                          https://repo.anaconda.com/pkgs/r/osx-64
                          https://repo.anaconda.com/pkgs/r/noarch
          package cache : /usr/local/anaconda3/pkgs
                          /Users/yren/.conda/pkgs
       envs directories : /usr/local/anaconda3/envs
                          /Users/yren/.conda/envs
               platform : osx-64
             user-agent : conda/22.11.1 requests/2.28.1 CPython/3.8.13 Darwin/22.2.0 OSX/10.16
                UID:GID : 504:20
             netrc file : /Users/yren/.netrc
           offline mode : False
```

```
 ᐅ conda env create -n cq-dev -f environment.yml
Collecting package metadata (repodata.json): done
Solving environment: done

Downloading and Extracting Packages

Preparing transaction: done
Verifying transaction: done
Executing transaction: done
Installing pip dependencies: | Ran pip subprocess with arguments:
['/usr/local/anaconda3/envs/cq-dev/bin/python', '-m', 'pip', 'install', '-U', '-r', '/Users/yren/github/CadQuery/cadquery/condaenv.d6a2kmgb.requirements.txt', '--exists-action=b']
Pip subprocess output:
Obtaining file:///Users/yren/github/CadQuery/cadquery (from -r /Users/yren/github/CadQuery/cadquery/condaenv.d6a2kmgb.requirements.txt (line 2))
  Preparing metadata (setup.py): started
  Preparing metadata (setup.py): finished with status 'done'
Collecting sphinxcadquery
  Downloading sphinxcadquery-1.3.3.tar.gz (264 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 264.4/264.4 kB 5.2 MB/s eta 0:00:00
  Preparing metadata (setup.py): started
  Preparing metadata (setup.py): finished with status 'done'
Collecting multimethod<2.0,>=1.7
  Downloading multimethod-1.9.1-py3-none-any.whl (10 kB)

Pip subprocess error:
ERROR: Ignored the following versions that require a different python version: 0.7 Requires-Python ~=2.7, >=3.4
ERROR: Could not find a version that satisfies the requirement cadquery-ocp<7.7,>=7.6 (from cadquery) (from versions: 7.5.3, 7.6.3a0)
ERROR: No matching distribution found for cadquery-ocp<7.7,>=7.6

failed

CondaEnvException: Pip failed
```

adam-urbanczyk, 2023-01-14:
Yup, confirmed: https://github.com/conda/conda/issues/6805

Used to work, but not anymore.

closed 2023-01-14 by jmwright

YHRen, 2023-01-15:
@adam-urbanczyk  thanks. the update works.