CadQuery/cadquery #1236: Cannot install in conda env anymore on an M1 mac

bernhard-42, 2023-01-15:
The latest commit prevents from dependencies being installed when in conda env.

On a M1 mac I have my own OCP wheel sitting on a private pypi server, called "banach"

My install process was for an existing conda env:

```bash
mamba create -n cq-3.9 python=3.9 ipython jupyterlab matplotlib numpy pandas     # for example, plus any other libs
pip install --pre --trusted-host banach --extra-index-url http://banach:8080/ git+https://github.com/CadQuery/cadquery.git
```

This now doesn't install any dependencies any more.

Why do I start with conda? Because mambaforge provides M1 arm64 packages that still don't exist sometimes on pypi.

If I fall back to the commit before the change in `setup.py` it works again

```bash
pip install --pre --trusted-host banach --extra-index-url http://banach:8080/ git+https://github.com/CadQuery/cadquery.git@4745ad1cee78923ff38992acc7a4177c37e5f5fe
```bash

This now installs everything from pypi except `cadquery_ocp` which comes from my private pypi server.

Is there a way to still get cadquery installed into a conda env via pip?

adam-urbanczyk, 2023-01-15:
Weird use case. If you can build a wheel, you can also build a conda package and install the deps from the env file. BTW: there is a PR in progress for arm64 OCP for the ocp-feedstock

If  you really need, maybe this will work:
```
env -u CONDA_PREFIX_1 pip install ... 
```

bernhard-42, 2023-01-15:
What do you mean with "weird use case". I and some others do have arm64 Macs and want to work with CadQuery. I don't see the weird part of this use case.

The wheel I built might have some issues (and I also had to create wheels for nlopt and casadi for arm64). While they do work with some simple tests, I don't want to publish them - and I have no clue about conda packaging. The approach up to the last commit was a good workaround until the OCP package exists.

And agreed, as soon as there is an OCP arm64 conda package it is a non issue any more. Nevertheless, until then a workaround is needed.

Maybe this is anyhow a good time to leave the conda ecosystem and move over to poetry ....

I'll give your suggestion a try tomorrow, thanks.

jmwright, 2023-01-16:
@bernhard-42 You can probably create your own local copy of the `environment.yml` file (and maybe the `setup.py` file too) which reverts the changes that were made in [this commit](https://github.com/CadQuery/cadquery/commit/a2c1f7c4a6176ef431c1dd84170d2caebef71e8e). Then you can install the conda environment with `conda env create -f environment_bernhard.yml -n cadquery`. That should create a conda environment named *cadquery* that will be the same as it was before the change so you can use your custom wheel.

adam-urbanczyk, 2023-01-16:
> What do you mean with "weird use case". I and some others do have arm64 Macs and want to work with CadQuery. I don't see the weird part of this use case.
> 

"Weird" was about mixing package managers and not the platform itself. BTW both casadi and nlopt are avaiable on conda-forge for osx-arm64.

bernhard-42, 2023-01-16:
Got it, thanks @jmwright and @adam-urbanczyk 

Not happy with mixing myself. However, I only had a private arm64 wheel for  cadquery_ocp (otherwise I couldn't work with cadquery on my machine).
If I now choose the pip route, I don't find arm64 binaries for casadi and nlopt. 
And if I go the conda route, I get casadi and nlopt, but no arm64 OCP.

And yes I could maintain my own environment.yml ...

Btw. what is the reason for preventing dependencies being installed via pip in a conda env. Usually, this will be controlled on the client side by ˋpip install --no-depsˋ and not prevented in the package, if I understand the python packaging approach correctly.

adam-urbanczyk, 2023-01-17:
See #1228 and https://github.com/pypa/pip/issues/9948

bernhard-42, 2023-01-17:
makes sense, thanks

adam-urbanczyk, 2023-01-20:
@bernhard-42  would you be able to test this experimental OCP build https://anaconda.org/cctbx-dev/ocp with CQ on your arm64 mac? 

See https://github.com/conda-forge/ocp-feedstock/pull/23 for context.

bernhard-42, 2023-01-20:
Sure, will test it tomorrow and let you know

bernhard-42, 2023-01-21:
@adam-urbanczyk I tried to make some kind of end-to-end test with vtk 9.2.2 - seems to work:

[image]

bernhard-42, 2023-01-21:
For reference, the environment setup:

```shell
mamba create -n ocp-test python=3.8
mamba activate ocp-test
mamba install -c cctbx-dev ocp vtk pyvista panel jupyterlab
```

adam-urbanczyk, 2023-01-23:
Thanks @bernhard-42 ! You can use osx-arm64 ocp from conda-forge now.

closed 2023-01-23 by adam-urbanczyk

bernhard-42, 2023-01-24:
Awesome! Thank you