CadQuery/cadquery #1242: RFC: context managers

hiaselhans, 2023-01-21:
Thanks for the great package! Once familiar with the api things work great!

However coming from python, the api feels a little counter-intuitive with things being on the stack or not and daisy chaining looking weird in python.

Here is an idea for an alternative interface to workplanes:

``` python
with cq.Workplane("XY") as wp:
    wp.moveTo(2,2)
    wp.circle(5)

element1 = wp.extrude(6)
```

so, instead of returning a new workplane with objects added to the stack, the workplane is mutable within the context-manager and chaining is cleaner

jmwright, 2023-01-21:
There has been quite a bit of discussion about a direct API over the years.  https://github.com/CadQuery/cadquery/wiki/Direct-API-Proposal

hiaselhans, 2023-02-10:
Thanks for the message @jmwright 

I read through the proposal and i do see a lot of value in there.
Since there has been quite a bit of discussion, can you tell me a bit more about the current state of it? Is the proposal something that is planned to be implemented in the near future? Why not breaking the current fluent API and start clean for a new version?

To me the most confusing thing in cadquery is that basically everything is a "Workplane" and every workplane has a lot of hidden context. When i extrude something i expect the result to be an "Object". Calling faces() i would expect a List[Face] or similar. instead i always get a Workplane with those objects or faces in the stack. It's quite hard to tell what is currently on the stack and what's not, while on the other hand IDE autocomplete would really speed things up with discrete types.

For a start, would the cadquery team accept a PR implementing the proposed API from this issue?  It would basically add a "mutable" boolean to the workplane which would add to it's own context instead of returning a new workplane.

adam-urbanczyk, 2023-02-10:
> Why not breaking the current fluent API and start clean for a new version?

Hard pass.

> 
> For a start, would the cadquery team accept a PR implementing the proposed API from this issue? It would basically add a "mutable" boolean to the workplane which would add to it's own context instead of returning a new workplane.

Take a look at `cq.Shape`. Otherwise, you can start a project that depends on CQ, the interfaces are quite stable.

hiaselhans, 2023-02-10:
> > Why not breaking the current fluent API and start clean for a new version?
> 
> Hard pass.
> 

Just an open question asked.
After reading the document, i have a hard time understanding the API roadmap:

- title: Direct API Proposal
- goal: new fluent api
- side goal: keep the old fluent api

and then there is also this:
https://cadquery.readthedocs.io/en/latest/primer.html#the-direct-api
So, probably an "old direct api" - before proposal?

Please don't get me wrong and excuse my dumb questions, but i am a little confused now.
I had a simple idea which in my opinion could make the interface to sketching a little cleaner and idiomatic but from the context provided i can't seem to find out if that's in any way wanted and in line with where the API is heading.

I really appreciate all the work you invest and I do like the code-first approach but i hoped there are ways to make it feel more intuitive and "fluid".

> Take a look at `cq.Shape`. Otherwise, you can start a project that depends on CQ, the interfaces are quite stable.

I think i should take that as a hard pass too and just implement my own private class?

jmwright, 2023-02-11:
@hiaselhans That API document was still under discussion and was evolving if I remember correctly. I think that @dcowden was leading that discussion.

It doesn't address your core questions, but if you want a list of faces, you can use `all()`. It's an extra step, but you can always get the lower level objects from the `Workplane` object.

```python
box = cq.Workplane.box(10, 10, 10)
print(len(box.faces().all()))
```

Jojain, 2023-02-11:
@hiaselhans 
As Jeremy already said there has been discussion about that. The conclusion was that the current API couldn't evolve into something like you propose without breaking everything, that's something that could only happen in a CQ 3.0 update.

The project build123d started as a sandbox for exploring such an API update. It has now evolved into its own thing but I think it matches what you are thinking of, so you might want to get a look. 

I too think that such an API update would be great but that requires a lot of work which could be spend into solving bugs or adding new features.

hiaselhans, 2023-02-11:
@Jojain Thanks a lot for the hint! It looks very much like what i was thinking of :)

@jmwright Thanks for the clarification!

closed 2023-02-11 by hiaselhans

adam-urbanczyk, 2023-02-11:
@hiaselhans could you explain what is missing in the `cq.Shape` API from your point of view?

hiaselhans, 2023-02-13:
Hi @adam-urbanczyk 

I don't think it's missing anything. It looks like an implementation of an cad entity.
But then again, looking at geomType() it looks like a very implicit type that in my opinion is a bit confusing.
In general i was looking for feedback on a builder-api based on context managers and thanks to @jojain i was pointed to a project addressing just that.