CadQuery/cadquery #1244: Issue with text

cyberhuman, 2023-01-23:
Hello. I have an issue with some characters with some fonts. It looks like the glyphs are not fused properly, which leads to problems when trying to etch some text on a solid.

Some of the character's shape is missing:
[image]
```py
import cadquery as cq

result = (
    cq.Workplane("XY")
        .box(2, 20, 20)
        .faces(">X")
        .workplane(centerOption = 'CenterOfBoundBox')
        .text("A", 18, -1, font = "Source Code Variable")
)
```

The whole face disappears:
[image]
```py
import cadquery as cq

result = (
    cq.Workplane("XY")
        .box(2, 20, 20)
        .faces("<X")
        .workplane(centerOption = 'CenterOfBoundBox')
        .text("A", 18, -1, font = "Source Code Variable")
)
```

It might be an issue with OCCT but I've been advised on Discord that it might be possible to work around it in CQ instead.

lorenzncode, 2023-01-24:
I can reproduce the issue.  The font is here: https://github.com/adobe-fonts/source-code-pro/releases
Select the VAR*.zip.

Experimenting with fusing the faces followed by clean resulted in a valid shape. 

https://github.com/CadQuery/cadquery/blob/4a79b31daefa7f44a758483638743111d0945783/cadquery/occ_impl/shapes.py#L3491

Changing to:

```py
    faces = Shape(builder.Perform(font_i, NCollection_Utf8String(text))).Faces()
    text_flat = faces[0]
    if len(faces) > 1:
        text_flat = text_flat.fuse(*faces[1:]).clean()

```

[image]

lorenzncode, 2023-01-25:
FontForge finds several characters with intersecting paths including: #$&,469;AKQRefk

[image]

Compare with the font used in the CQ tests where no glyphs have intersecting paths.  

It may be better to fix this font for use with CQ/OCCT.  It seems easy enough with FontForge.

1. Open the font in FontForge

  ```
  $ fontforge SourceCodeVariable-Roman.ttf
  ```

2. Select the characters of interest

  [image]

3. Remove Overlap
    Element->Overlap->Remove Overlap

4. Save the file
    File->Generate Fonts...
    Generate
 
I selected TrueType, then specified a local fontPath.:

```py
text("A", 18, -1, font="Source Code Variable", fontPath ="./SourceCodeVariable-Roman-clean.ttf")
```
[SourceCodeVariable-Roman-clean.zip](https://github.com/CadQuery/cadquery/files/10495438/SourceCodeVariable-Roman-clean.zip)

cyberhuman, 2023-01-25:
Thanks! It's a nice observation, but think it's not a good solution at all. IMO it should not matter for CQ if the font used has this issue or not, as it works in every other software.

adam-urbanczyk, 2023-01-25:
The underlying kernel does not handle self-intersections. Just make sure that you use well-formed fonts (or fix them in [fontforge](http://designwithfontforge.com/en-US/Making_Sure_Your_Font_Works_Validation.html)). I'm not a font nerd, but it seems that Blender and Godot have similar issue, so such a requirement is not that exotic.

closed 2023-01-25 by adam-urbanczyk