CadQuery/cadquery #1248: Can't pip install Cadquery 2.x.x with python version 3.11.x

medicationforall, 2023-01-25:
## To Reproduce
* I have python version **3.11.1** installed on a windows machine
* when I run pip install --pre cadquery
* the version of cadquery installed is 1.2.5

Installing a library that requires a 2.x.x  cadquery version I get a stacktrace.

``` bash
ERROR: Ignored the following versions that require a different python version: 2.2.0b0 Requires-Python >=3.8,<3.11; 2.2.0b1 Requires-Python >=3.8,<3.11; 2.2.0b2 Requires-Python >=3.8,<3.11

ERROR: Could not find a version that satisfies the requirement cadquery==2.2.0b2 (from skirmishbunker) (from versions: 0.1.0, 0.1.1, 0.1.2, 0.1.3, 0.1.4, 0.1.5, 0.1.7, 0.1.8, 0.2.0, 0.3.0, 0.4.0, 0.4.185, 0.4.186, 0.5.1, 0.5.2, 1.0.0, 1.1.0, 1.2.0, 1.2.1, 1.2.2, 1.2.4, 1.2.5)

ERROR: No matching distribution found for cadquery==2.2.0b2
```

## Environment

OS: Windows 11
Was CadQuery installed using Conda?: No I was using pip and a virtual environment.
Using:  command line

## Notes
Installing cadquery version 2.2.0b2  with python version 3.10.6 on my other laptop worked fine.

jmwright, 2023-01-25:
We're working on Python 3.11 support now, but it's not ready yet.

medicationforall, 2023-01-25:
> We're working on Python 3.11 support now, but it's not ready yet.

All good!, I don't claim to know the underlying requirements.

jmwright, 2023-01-31:
Python 3.11 should be working in conda now, and we're still waiting on one of our dependencies (casadi) to have a Python 3.11 wheel available before 3.11 is working via pip/PyPI.

peterdsharpe, 2023-05-18:
Hi @jmwright ,

CasADi now has 3.11 support: https://pypi.org/project/casadi/3.6.3/#files

Is it possible to release a CadQuery update to accomodate 3.11? [I have a downstream package](https://github.com/peterdsharpe/AeroSandbox) that incidentally uses both CasADi and CadQuery, and once CadQuery provides 3.11 support, I can also provide 3.11 support :)

jmwright, 2023-05-18:
@peterdsharpe Does it not work for you? I just installed cadquery-ocp in Ubuntu 23.04 (Python 3.11) and it works fine. I could be wrong, but I think it should just pick up on the fact that there is a Casadi wheel for Python 3.11 now.

peterdsharpe, 2023-05-18:
As of [this CI run](https://github.com/peterdsharpe/AeroSandbox/actions/runs/5015432204) around 4 hours ago, using Python 3.11 on Ubuntu, `pip` does not detect any `cadquery` versions that satisfy requirements.

Here's the relevant section of the error that occurs during installation:
```
INFO: pip is looking at multiple versions of aerosandbox[full,test] to determine which version is compatible with other requirements. This could take a while.
ERROR: Ignored the following versions that require a different python version: 2.2.0 Requires-Python >=3.8,<3.11; 2.2.0b0 Requires-Python >=3.8,<3.11; 2.2.0b1 Requires-Python >=3.8,<3.11; 2.2.0b2 Requires-Python >=3.8,<3.11
ERROR: Could not find a version that satisfies the requirement cadquery>=2; python_version >= "3.8" and extra == "full" (from aerosandbox[full,test]) (from versions: 0.1.0, 0.1.1, 0.1.2, 0.1.3, 0.1.4, 0.1.5, 0.1.7, 0.1.8, 0.2.0, 0.3.0, 0.4.0, 0.4.185, 0.4.186, 0.5.1, 0.5.2, 1.0.0, 1.1.0, 1.2.0, 1.2.1, 1.2.2, 1.2.4, 1.2.5)
ERROR: No matching distribution found for cadquery>=2; python_version >= "3.8" and extra == "full"
Error: Process completed with exit code 1.
```

So, it looks like the `python_version<3.11` dependency is hard-pinned in the latest (2.2.0) version of CadQuery on PyPI. (Even though it sounds like it should install without issue if the pin were removed, now that CasADi has 3.11 wheels.)

My understanding is that to remove the pin, one would have to release an updated version (a hotfix of some sort) of CadQuery on PyPI.

adam-urbanczyk, 2023-05-19:
If you don't need a release, you could pip install from git.

peterdsharpe, 2023-05-20:
@adam-urbanczyk That is true, and a good suggestion! 

I would prefer to avoid pinning a dependency of my package to a git repo, for a few reasons:
- Reproducibility - without having a SemVer-esque version to refer to, it becomes very difficult to make repeatable installations, if needed in order to track down a bug. If I instead pin it to a specific commit, my downstream users won't get new `cadquery` features as they're released, and there's also a possibility that I'm setting up a dependency on unstable or untested code. Also if the repository is deleted or moved, the link breaks.
- Ease of Use - Most of my package's users are in aerospace, where having git installed is not particularly common. Many are also behind corporate firewalls where installing directly from GitHub tends to have more problems than installing from PyPI.

It might be worthwhile to do a temporary pip install from git in my development environment until the cadquery package is updated on PyPI. But, for the distributed version of my package, it's preferable to depend on stable releases on PyPI for the reasons mentioned above.

adam-urbanczyk, 2023-05-21:
OK, clear. @jmwright are you ok with releasing 2.3 or 2.2.1?

jmwright, 2023-05-21:
I think the original plan was to start doing a January and June releases each year anyway, so I think we could just release 2.3.0.

I created a doc in January on [how to do a release](https://github.com/CadQuery/cadquery/wiki/Developer-Doc:-Creating-a-New-Release). You or @lorenzncode could try to follow that doc to see how complete it is.

mojca, 2023-06-18:
Here's another request from me to release a package that's compatible with Python 3.11.
Even if the version on pypi is compatible, it refuses to install due to explicitly declared incompatibility with Python 3.11.

mojca, 2023-06-18:
I tried installing it on Windows with `--ignore-requires-python`, but it doesn't really help because it doesn't find a compatible `cadquery-ocp` version.
```
$ pip install --ignore-requires-python  cadquery[dev]==2.2.0b2
Collecting cadquery[dev]==2.2.0b2
  Using cached cadquery-2.2.0b2-py3-none-any.whl (142 kB)
INFO: pip is looking at multiple versions of cadquery[dev] to determine which version is compatible with other requirements. This could take a while.
ERROR: Could not find a version that satisfies the requirement cadquery-ocp<7.7,>=7.6 (from cadquery[dev]) (from versions: 7.7.0a0, 7.7.0)
ERROR: No matching distribution found for cadquery-ocp<7.7,>=7.6
```

mojca, 2023-06-18:
Unrelated to this ticket, but I tried installing on Ubuntu 22.04 inside WSL. Installation worked, but executing the code failed:
```
RuntimeError: module compiled against API version 0xf but this version of numpy is 0xe
Traceback (most recent call last):
  File "/path/to/model.py", line 1, in <module>
    import cadquery as cq
  File "/usr/local/lib/python3.10/dist-packages/cadquery/__init__.py", line 22, in <module>
    from .occ_impl import exporters
  File "/usr/local/lib/python3.10/dist-packages/cadquery/occ_impl/exporters/__init__.py", line 10, in <module>
    from ...cq import Workplane
  File "/usr/local/lib/python3.10/dist-packages/cadquery/cq.py", line 61, in <module>
    from .sketch import Sketch
  File "/usr/local/lib/python3.10/dist-packages/cadquery/sketch.py", line 28, in <module>
    from .occ_impl.sketch_solver import (
  File "/usr/local/lib/python3.10/dist-packages/cadquery/occ_impl/sketch_solver.py", line 10, in <module>
    import nlopt
  File "/usr/local/lib/python3.10/dist-packages/nlopt/__init__.py", line 1, in <module>
    from .nlopt import *
  File "/usr/local/lib/python3.10/dist-packages/nlopt/nlopt.py", line 20, in <module>
    from . import _nlopt
ImportError: numpy.core.multiarray failed to import
```

jmwright, 2023-06-25:
@adam-urbanczyk I can do release 2.3.0 this week. Are there any open PRs that we want to merge before that release?

adam-urbanczyk, 2023-06-25:
Sounds good, no PRs to be merged before.

jmwright, 2023-06-26:
PR for the 2.3.0 release is available [here](https://github.com/CadQuery/cadquery/pull/1355). This release will include the removal of the upper Python version restriction.

peterdsharpe, 2023-06-29:
Thanks for the release, @jmwright !

Unfortunately I'm still seeing install issues with CadQuery 2.3.1 on Python 3.11 (on Ubuntu, via `pip`), albeit different ones now - the issue appears to be coming from `nlopt`.

[Here is the relevant section of a logfile for a GitHub Actions CI build for a project, attempting to install CadQuery 2.3.1 on Python 3.11](https://github.com/CadQuery/cadquery/files/11909524/6_Test.with.pytest.with.extras.only.tutorials.txt). This logfile is a section of [this CI build](https://github.com/peterdsharpe/AeroSandbox/actions/runs/5415529021/jobs/9843975612).

This does, of course, suggest a possible downstream issue in `nlopt` that should be resolved there - but I wanted to ask if anyone else (@jmwright , @mojca , etc.) has been able to get CadQuery 2.3.1 to install on Python 3.11, on Ubuntu, using `pip`.

mojca, 2023-06-29:
I tried installing the older version two weeks ago (Ubuntu 22.04) and it broke at one of dependencies at runtime due to incompatible numpy version.

jmwright, 2023-06-29:
I get the same nlopt error.

mojca, 2023-06-29:
It doesn't work on Windows either:
```
Traceback (most recent call last):
  File "C:\Program Files\Python311\Lib\site-packages\nlopt\nlopt.py", line 22, in swig_import_helper
    return importlib.import_module(mname)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Program Files\Python311\Lib\importlib\__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1206, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1178, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1149, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 676, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 573, in module_from_spec
  File "<frozen importlib._bootstrap_external>", line 1233, in create_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
ImportError: DLL load failed while importing _nlopt: The specified module could not be found.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "D:\test.py", line 1, in <module>
    import cadquery as cq
  File "C:\Program Files\Python311\Lib\site-packages\cadquery\__init__.py", line 22, in <module>
    from .occ_impl import exporters
  File "C:\Program Files\Python311\Lib\site-packages\cadquery\occ_impl\exporters\__init__.py", line 10, in <module>
    from ...cq import Workplane
  File "C:\Program Files\Python311\Lib\site-packages\cadquery\cq.py", line 61, in <module>
    from .sketch import Sketch
  File "C:\Program Files\Python311\Lib\site-packages\cadquery\sketch.py", line 28, in <module>
    from .occ_impl.sketch_solver import (
  File "C:\Program Files\Python311\Lib\site-packages\cadquery\occ_impl\sketch_solver.py", line 10, in <module>
    import nlopt
  File "C:\Program Files\Python311\Lib\site-packages\nlopt\__init__.py", line 1, in <module>
    from .nlopt import *
  File "C:\Program Files\Python311\Lib\site-packages\nlopt\nlopt.py", line 25, in <module>
    _nlopt = swig_import_helper()
             ^^^^^^^^^^^^^^^^^^^^
  File "C:\Program Files\Python311\Lib\site-packages\nlopt\nlopt.py", line 24, in swig_import_helper
    return importlib.import_module('_nlopt')
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Program Files\Python311\Lib\importlib\__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
ModuleNotFoundError: No module named '_nlopt'
```

jmwright, 2023-06-30:
nlopt Python 3.11 issue: https://github.com/DanielBok/nlopt-python/issues/19

adam-urbanczyk, 2023-08-06:
Nlopt issue is resolved, can I close this?

jmwright, 2023-08-06:
Yes, I just confirmed that the CadQuery installation works fine now with Python 3.11.

closed 2023-08-06 by jmwright