CadQuery/cadquery #1249: Flatten STEP File Structure

jmwright, 2023-01-25:
The KiCAD project has existing STEP files that have a very flat structure. The following screenshot shows the structure when the file is imported.

[image]

The reason they want this is because they can have full board models with thousands of components and they don't want any extra structure to deal with. Their STEPS do include the color, which is a requirement.

I've spent some time doing experiments with exporting to STEP, with some similarities to how we do it currently ([gist here](https://gist.github.com/jmwright/f541f2bf78697a1f157349ccc4a63351)). I have tried different ways of structuring the exported shapes including making a compound, and adding a shape and adding all the faces to that. No matter what I do the structure of the STEP looks like this when imported into FreeCAD.

[image]

Am I missing an option somewhere to flatten the structure? Is there a way to accomplish this? Any ideas @adam-urbanczyk ?

jmwright, 2023-01-26:
The first screenshot only seems to be possible if the bounds of the assembly components are intersecting with one another. The original KiCAD 3D files were exported on FreeCAD, and (using FreeCAD) if I have the components intersect I can get the structure from the first screenshot. If they don't intersect, I get something similar to the structure in the second screenshot. I'm going to close this issue as it seems that the second screenshot will be as good as it's going to get with properly constrained assemblies.

adam-urbanczyk, 2023-01-27:
I'm not sure that I understand the request here. You want to get rid of the extra `boxes` and `origin` entities?

jmwright, 2023-01-27:
> Did you maybe mean single solid versus multiple solids?

Yes

>  If what FreeCAD does is the golden standard for this application, then simply take a look at the sources.

I have been. They appear to fuse the solids and then map the original faces to add the colors back in.

adam-urbanczyk, 2023-01-27:
> 
> I have been. They appear to fuse the solids and then map the original faces to add the colors back in.

Sounds reasonable, do you want to contribute this to CQ? It might be useful to other users.

adam-urbanczyk, 2023-01-27:
AFAICT this is the relevant file: https://github.com/FreeCAD/FreeCAD/blob/b44d70f6460b8175cc1c554d68b86b432845563b/src/Mod/Import/App/ImportOCAF2.cpp

jmwright, 2023-01-27:
> do you want to contribute this to CQ? It might be useful to other users.

I hope to once I have it working properly. Things get a little strange with the faces when they overlap in certain ways.  [This comment](https://github.com/CadQuery/cadquery/issues/23#issuecomment-444713864) from 2018 helps illustrate some of the strangeness.

I think we can also simplify our unfused STEP file structure. Right now the assemblies are more deeply nested. I think that we can collapse that down to the top level assembly and it's components, while still pulling name and color info from the assembly. This is the current structure:

[image]

And this is the structure when the cq.Assembly is handled as a STEP assembly with component entries in the STEP file. I have full control over the names and colors by setting them in the `cq.Assembly.Add()` calls. This is the result of the code that I have at the bottom of this comment.

[image]

So my initial thought is this: We keep the current STEP export, but add two other options. One is a flattened structure (like in the screenshot above), and the other is the fused structure with colors. The caveat with the fused-with-colors option is that strange things can happen with the colors in certain situations. I don't really have any ideas yet how to cleanly integrate these options into the current `Assembly.save()` method though.

Here is the sample code I wrote to accomplish the unfused STEP export. Note that I don't use `Assembly.toCompound()` in my code because it ends up adding extra nesting. I build the compound manually instead. I haven't investigated the reason for the extra `toCompound()` nesting yet, and don't really understand it.

```python
from OCP.TopoDS import TopoDS_Compound
from OCP.BRep import BRep_Tool, BRep_Builder
from OCP.Quantity import Quantity_Color, Quantity_TOC_RGB
from OCP.TCollection import TCollection_ExtendedString
from OCP.TDataStd import TDataStd_Name
from OCP.TDF import TDF_Label
from OCP.TDocStd import TDocStd_Application, TDocStd_Document
from OCP.XCAFDoc import XCAFDoc_DocumentTool, XCAFDoc_ColorType, XCAFDoc_ColorGen
from OCP.XSControl import XSControl_WorkSession
from OCP.STEPControl import STEPControl_StepModelType
from OCP.STEPCAFControl import STEPCAFControl_Writer
import cadquery as cq

def assy_to_shallow_step(assy):
    # Get set up for the STEP export
    app = TDocStd_Application()
    doc = TDocStd_Document(TCollection_ExtendedString("XmlOcaf"))
    app.InitDocument(doc)

    # Shape and color tools
    shape_tool = XCAFDoc_DocumentTool.ShapeTool_s(doc.Main())
    color_tool = XCAFDoc_DocumentTool.ColorTool_s(doc.Main())

    # Set up the writer
    session = XSControl_WorkSession()
    writer = STEPCAFControl_Writer(session, False)
    writer.SetColorMode(True)
    writer.SetLayerMode(True)
    writer.SetNameMode(True)

    # Set up the compound
    comp = TopoDS_Compound()
    comp_builder = BRep_Builder()
    comp_builder.MakeCompound(comp)
    for part in assy.children:
        # For simplicity in this example code I don't handle assembly structure properly
        comp_builder.Add(comp, part.obj.solids().all()[0].val().wrapped)

    # Add the top level shape, have it broken apart into an assembly and set the top-level name
    top_level_lbl = shape_tool.AddShape(comp, True)
    TDataStd_Name.Set_s(top_level_lbl, TCollection_ExtendedString("test_assembly"))

    # Assign names and colors based on the assembly
    for part in assy.children:
        cur_lbl = shape_tool.FindShape(part.obj.solids().all()[0].val().wrapped)
        TDataStd_Name.Set_s(cur_lbl, TCollection_ExtendedString(part.name))
        color_tool.SetColor(cur_lbl, part.color.wrapped, XCAFDoc_ColorGen)

    # Do the write of the STEP file to disk
    writer.Transfer(doc, STEPControl_StepModelType.STEPControl_AsIs)
    status = writer.Write("/home/jwright/Downloads/shallow_step_export_test.step")

# Build a simple assembly
assy = cq.Assembly()
box1 = cq.Workplane().box(10, 10, 10)
assy.add(box1, color=cq.Color(1, 0, 0), name="box1")
box2 = cq.Workplane().center(10, 10).box(10, 10, 10)
assy.add(box2, color=cq.Color(0, 1, 0), name="box2")

# Call the custom export method
assy_to_shallow_step(assy)

```

adam-urbanczyk, 2023-01-27:
OK, so it seems that to get an opaque step one needs to use a flat compound (i.e. without compounds within compound) and simply pass `False` when calling `AddShape`.

jmwright, 2023-02-02:
I have done a test integration with the KiCAD generator, and the following code seems to generate STEP files that meet all of their criteria (file size, STEP nesting structure, etc). It exports a fused compound that retains the colors. I have posted some samples to the [issue on GitLab](https://gitlab.com/kicad/libraries/kicad-packages3D-generator/-/issues/5#note_1263615122) to ask for verification.
```python
def assy_to_fused_step(assy, filepath, name):
    # Get set up for the STEP export
    app = TDocStd_Application()
    doc = TDocStd_Document(TCollection_ExtendedString("XmlOcaf"))
    app.InitDocument(doc)

    # Shape and color tools
    shape_tool = XCAFDoc_DocumentTool.ShapeTool_s(doc.Main())
    color_tool = XCAFDoc_DocumentTool.ColorTool_s(doc.Main())

    # Set up the STEP document writer
    session = XSControl_WorkSession()
    writer = STEPCAFControl_Writer(session, False)
    writer.SetColorMode(True)
    writer.SetLayerMode(False)
    writer.SetNameMode(True)
    Interface_Static.SetIVal_s("write.surfacecurve.mode", False)
    Interface_Static.SetIVal_s("write.precision.mode", False)

    # Fuse the assembly into a compound
    comp = assy.toCompound().fuse()

    # Add the top level shape
    top_level_lbl = shape_tool.AddShape(comp.wrapped, False)
    TDataStd_Name.Set_s(top_level_lbl, TCollection_ExtendedString(name))

    # Walk through all the components of the assembly
    i = 0
    for part in assy.children:
        other_subs = assy.children.copy()
        other_subs.pop(i)

        # Find the compound of this component that are its own within the fused solid
        diffed_sub = part.obj
        for other_sub in other_subs:
            diffed_sub = diffed_sub.cut(other_sub.obj)

        # Add the faces that are part of this assembly component
        for face in diffed_sub.faces().all():
            for comp_face in comp.Faces():
                # See if we have the face that needs to be worked with
                if face.val().Center() == comp_face.Center() and face.val()._uvBounds() == comp_face._uvBounds():
                    # Add the face as a subshape and set its color to match the parent assembly component
                    cur_lbl = shape_tool.AddSubShape(top_level_lbl, comp_face.wrapped)
                    color_tool.SetColor(cur_lbl, part.color.wrapped, XCAFDoc_ColorGen)

        i += 1

    # Write the resulting STEP document to disk
    writer.Transfer(doc, STEPControl_StepModelType.STEPControl_AsIs)
    status = writer.Write(filepath)
```

adam-urbanczyk, 2023-02-02:
Thanks for sharing. It looks like this solution will be quite slow, is it good enough for KiCAD?

jmwright, 2023-02-02:
> is it good enough for KiCAD?

I think so. I'm my tests it didn't significantly slow down the generator.