CadQuery/cadquery #1252: Version reported as 0.0.0 when installing with conda

lorenzncode, 2023-01-28:
```
>>> import cadquery
>>> cadquery.__version__
'0.0.0'
```

lorenzncode, 2023-01-28:
This is true at least for master.

Perhaps #1245 will correct the version for the release but then does it go back to 0.0.0 for master?

jmwright, 2023-01-28:
@lorenzncode Maybe it's related to the environment variable mentioned in #1085 ? Start [here](https://github.com/CadQuery/cadquery/pull/1085#issuecomment-1138674053) and ready the previous few comments to get the context.

lorenzncode, 2023-01-28:
@jmwright Thanks, I see it's set here with build variable `Build.SourceBranchName`.

https://github.com/CadQuery/conda-packages/blob/40f77ee43585a70e5958f6d7890e793337630ce1/conda-build.yml#L71

adam-urbanczyk, 2023-03-14:
This is not a "when installing with conda" issue, but rather "when installing a dev version" issue.

Release version for at least 2.2 reports `__version__` correctly.

closed 2023-04-04 by jmwright