CadQuery/cadquery #1263: DXF export spline location

sethfischer, 2023-02-06:
The DXF generated by the following has its splines offset. A bug appears to be in `cadquery.occ_impl.exporters.dxf._dxf_spline`.

```python
"""Export text as DXF."""

import cadquery as cq
from cadquery import exporters

text = cq.Workplane().text("cQ", 5, 1)

exporters.dxf.exportDXF(text, "output.dxf")
```
`output.dxf` viewed in LibreCad 2.1.3:

[image]

lorenzncode, 2023-02-18:
I experimented with this and a transformation may be required perhaps something like the following in `_dxf_spline`:
```spline.Transform(plane.rG.wrapped.Trsf()*adaptor.Trsf())```

and a similar transformation in the other dxf methods.

Testing with:
```
text = cq.Workplane("XZ").text("CQ", 5, 1)
cq.exporters.dxf.exportDXF(text.section(), "output.dxf")
```
[image]

adam-urbanczyk, 2023-03-16:
Solved in #1267

closed 2023-03-16 by adam-urbanczyk