CadQuery/cadquery #1266: Eliminate level of hierarchy Assembly export

lorenzncode, 2023-02-10:
Assembly STEP export currently creates a root node with hardcoded name "CQ assembly". 

[image]

(from #1239 code with unsupported constraint removed)

This root node could be considered extraneous and be removed.
[image]

Regarding the root name,  I think the existing naming can be used here - the user specifies a name else it is auto generated uuid.

jmwright, 2023-02-27:
@lorenzncode What is the advantage of using UUID over a default of "CQ assembly"? It's it just to avoid naming collisions, or are there other reasons?

What module/method should be used to generate the UUID?

lorenzncode, 2023-02-27:
@jmwright A name may optionally be provided when creating an assembly. 

```py
assy = cq.Assembly(name = "myassy")
```

If a name is not given, a UUID is automatically assigned as the name.
https://github.com/CadQuery/cadquery/blob/0aa889cdcb4598d22da304d94efc28befeecb423/cadquery/assembly.py#L118

PR #1270 removes the extraneous top level node that had been created on STEP export.  Now that the root node is an assembly, the assembly name itself can be used.  Changing the name to a different hard coded default would require additional logic.  I didn't see the value of adding special handling to change the name from the UUID to say "CQ Assembly" if a name was not given.

For PR #1261, could you access the existing Assembly `name` property rather than add a separate interface with `assembly_name`?

jmwright, 2023-02-28:
> For PR https://github.com/CadQuery/cadquery/pull/1261, could you access the existing Assembly name property rather than add a separate interface with assembly_name?

Yes, it should have been done that way but I forgot about being able to specify a name for the assembly. I'll work on making that change and using a UUID for when the name is not specified.

PR #1270 (2023-03-08): Remove one level of hierarchy in STEP export
Do not hard code root name if unspecified

Resolves #1266
closed 2023-03-08 by jmwright