CadQuery/cadquery #1272: text cut with custom font breaks face

jniclas, 2023-02-13:
Hey,

I tried to cut the text in a solid, using a custom font, but the upper face breaks. I tried three fonts: 

- Quicksand-Bold.ttf
- Comfortaa-Bold-tff
- Nunito.ttf

All have the same effect:

[image]

With default font it works fine:

[image]

I used the following code:

```python
from jupyter_cadquery import (
    Edges,
    Faces,
    Part,
    PartGroup,
    Vertices,
    close_viewer,
    close_viewers,
    get_defaults,
    get_viewer,
    open_viewer,
    set_defaults,
    show,
)
from jupyter_cadquery.replay import disable_replay, enable_replay, replay, reset_replay

import cadquery as cq
from cadquery import exporters
import math
import os
set_defaults(axes=True, timeit=False, control="orbit")
cv = open_viewer("Examples", cad_width=640, height=480, glass=True)
enable_replay(False, False)
show_object = replay

circle_outer_end = 10
length = 30
thickness = 2
fillet = 0.6
text = "CQ"

c = length

directory = os.getcwd()

result = (
    cq.Workplane("XY")
    .moveTo(c, 0)
    .circle(circle_outer_end)
    .extrude(thickness, combine=True)
    .edges()
    .fillet(fillet)
    .faces(">Z")
    .workplane()
    .center(c, 0)
    .transformed(rotate=(0, 0, -90))
    .text(
        text,
        fontsize=8,
        distance=-0.5,
        fontPath=os.path.join(directory, "phijn_font/Quicksand-Bold.ttf"),
        cut=False,
        kind="bold",
        combine="cut",
        clean=True
    )
)

show(result)
```

On closer inspection I have the feeling the font is not properly converted to a solid:

[image]

Can I do something on my side?

adam-urbanczyk, 2023-02-13:
Try taking a look here #1244 (fontforge part).

jniclas, 2023-02-14:
I tried with FontForge, did not work sadly. 
But afterwards I had overwritten the method Compound.makeText like shown in the comment https://github.com/CadQuery/cadquery/issues/1244#issuecomment-1401193688 and that worked great for Comfortaa-Bold.tff and Nunito-Bold.tff!   ...But not for Quicksand-Bold.tff 
Is there anything against it to integrate the three changed lines to CadQuery? Would at least increase the number of usable fonts.
[image]