CadQuery/cadquery #1273: Selection of diagonal faces

mukral, 2023-02-14:
Hi,I'm struggling to select a faces that are not parallel to one of the axis like in all examples.
In my example below I've tried to find the upper face on the box, because so I've select the farthest X and Z edges.
But it selects the surface which is below. Can anybody please give me an example howto select the upper diagonal surface.
Sorry, I've really read a lot docu and tried many different ways, but didn't succeded.
```
import cadquery as cq
box1 = cq.Workplane().transformed(offset=cq.Vector(0, 0,0),rotate=cq.Vector(0, 45)).sketch().rect(10, 10).finalize().extrude(5)
cylinder1 = box1.faces(">X and >Z").workplane().circle(5).extrude(7)
show_object(box1)
show_object(cylinder1)
```
Thank you very much
Markus

adam-urbanczyk, 2023-02-14:
Try `box1.faces('>(1,0,1)')`

mukral, 2023-02-14:
Hi, depending to this: I would like to translate the cylinder in direction to the z-axis of the "new" rotated workplane-system (to get it at the center of box-surface), but otherwise then the extrusion-direction, the translation-direction seems to be bound to the absolute system and went straight upwards. Did there an option that I didn't set?
``` 
import cadquery as cq
box1 = cq.Workplane().transformed(offset=cq.Vector(0, 0,0),rotate=cq.Vector(0, 45)).sketch().rect(10, 10).finalize().extrude(5)
cylinder1 = box1.faces(">X and >Z").workplane().circle(5).extrude(7).translate((0,0,5))
show_object(box1)
show_object(cylinder1)
```

Sorry, I really try to get solutions on my own but nothing worked.
Thank you,
Markus

mukral, 2023-02-14:
Adam, thank you so much - another door opens :-)

mukral, 2023-02-15:
For other people with similar challenges - a mixture of all kind of selectors and combinations may help to find the correct face.
f.i. when I stretched the box, the workplane switched, It needed some time but the combination below helped.
```
import cadquery as cq
#box1 = cq.Workplane().box(10,10,5).rotate((0,0,0),(0,1,0),45)
box1 = cq.Workplane().transformed(offset=cq.Vector(0,0,0),rotate=cq.Vector(0, 30)).sketch().rect(30, 10).finalize().extrude(5).translate((0,0,0))
cylinder1 = box1.faces(">(1,0,1) and >Z").workplane().circle(5).extrude(7)
show_object(box1)
show_object(cylinder1)
```
Markus

mukral, 2023-02-15:
oh no, did not help - it's not the upper surface but the lower one again - what a fight...

lorenzncode, 2023-02-16:
After changing the rotation from 45 to 30 degrees the selector `>(1, 0, 1)` will no longer work.

Here is an example using `DirectionSelector`.

```py
import cadquery as cq

box1 = (
    cq.Workplane()
    #.transformed(offset=(0, 0, 0), rotate=(0, 45))
    .transformed(offset=(0, 0, 0), rotate=(0, 30))
    .sketch()
    .rect(30, 10)
    .finalize()
    .extrude(5)
)

# to visualize the selected face
theface = box1.faces(cq.DirectionSelector(box1.plane.zDir))

# works for rotate=(0, 45)
# cylinder1 = box1.faces(">(1, 0, 1)").workplane().circle(5).extrude(7)

cylinder1 = (
    box1.faces(cq.DirectionSelector(box1.plane.zDir))
    .workplane()
    .circle(5)
    .extrude(7, combine=False)
)

show_object(box1, name="box1")
show_object(theface, name="theface")
show_object(cylinder1, name="cylinder1")
```

mukral, 2023-02-16:
ah - I see, DirectionSelector is here the instrument to switch from the "normal/absolut" reference system (where 45 degree face is >(1,0,1) and 30 degree is not) to "relative to something you want to be bound to" so you can find planes in that direction... thanks a lot.

Markus

closed 2023-02-16 by mukral