CadQuery/cadquery #1276: extrude until next/last fails with the hollow profiles

bao-eng, 2023-02-25:
I encountered a problem with extruding the hollow profiles until the existing face.
There is the following test for extrude until with multiple distant wires:
https://github.com/CadQuery/cadquery/blob/4c2a0383495862c71aaf90da2240dea9d40f99bf/tests/test_cadquery.py#L3345
The following test modification will produce an incorrect result:
```python
from cadquery import *
wp_ref = Workplane("XY").box(10, 10, 10).center(20, 0).box(10, 10, 10)
wp = (
    wp_ref.faces(">X[1]")
    .workplane(centerOption="CenterOfMass")
    .rect(2, 2)
    .rect(4, 4)
    .extrude("next")
)
```
[image]
The following test modification will fail with ```Standard OutOfRange: NCollection_Sequence:Value```:
```python
from cadquery import *
wp_ref = Workplane("XY").box(10, 10, 10).center(20, 0).box(10, 10, 10)
wp = (
    wp_ref.faces(">X[1]")
    .workplane(centerOption="CenterOfMass")
    .circle(2)
    .circle(4)
    .extrude("next")
)
```

Jojain, 2023-02-25:
Yes it is a limitation you cannot have a section with a hole with "next" and "last" options.

bao-eng, 2023-03-05:
> 

Thank you, as a workaround, I used a quarter profile that does not have an opening.