CadQuery/cadquery #1279: Non planar circles exported to DXF in same plane

lorenzncode, 2023-03-04:
```py
import cadquery as cq

c1 = cq.Wire.makeCircle(2, (0, 0, 0), (0, 0, 1))
c2 = cq.Wire.makeCircle(2, (4, 0, 0), (0, 1, 1))

result = cq.Workplane("XY").add([c1, c2])

cq.exporters.export(result, "test_circle.dxf", "DXF")
```

I think the current behavior may be unexpected.  Perhaps the two circles should be exported to DXF as they are with different normals, or raise an error when the circle is not in the same plane as the workplane.

lorenzncode, 2023-03-05:
I suppose this code breaks the assumptions and goes against the purpose of DXF in CadQuery.  To preserve the 3D shapes I could export to STEP.  DXF is used to import 2D profiles and export to 2D.

closed 2023-03-05 by lorenzncode

adam-urbanczyk, 2023-03-05:
Indeed, only 2D is supported.