CadQuery/cadquery #1282: Export SVG in Millimeter

Paul8043, 2023-03-10:
I am trying to port some models from OpenSCAD to CadQuery. The goal ist to create a 2D-SVG-file which can be used by a ShaperOrigin. The  ShaperOrigin expects that all measures are in Millimeters (at least in my country). The SVG-file produced by CadQuery seems to use pixels as unit, which does not fit for my application. I couldn't find any options to force the unit used by the exporter to be Millimeters. I would also like to know what is the correct way to disable the "projectionDir" for 2D output.

Please help. This is a functionality needed for further migrations.

Here is a small code example which shows the problem:

```
import cadquery as cq
from cadquery import exporters
part = cq.Workplane("XY").rect(100,100,centered=False)
show_object(part,name="part",options={"alpha":0.2,"color":(255,170,0)})

exporters.export(
            part,
            'cadquery_output.svg',
            opt={
                "width": 100,
                "height": 100,
                "marginLeft": 0,
                "marginTop": 0,
                "showAxes": False,
                "projectionDir": (0.0,0.0,1.0),
                "strokeWidth": 1.0,
                "strokeColor": (255, 0, 0),
                "hiddenColor": (0, 0, 255),
                "showHidden": False,
            },
        )
```

The content of the SVG-File shows a square which is much too small:

```
<?xml version="1.0" encoding="UTF-8" standalone="no"?>
<svg
   xmlns:svg="http://www.w3.org/2000/svg"
   xmlns="http://www.w3.org/2000/svg"
   width="100.0"
   height="100.0"

>
    <g transform="scale(0.75, -0.75)   translate(0.0,-100.0)" stroke-width="1.0"  fill="none">
       <!-- hidden lines -->
       <g  stroke="rgb(0,0,255)" fill="none" stroke-dasharray="1.0,1.0" >

       </g>

       <!-- solid lines -->
       <g  stroke="rgb(255,0,0)" fill="none">
			<path d="M0.0,0.0 L100.0,0.0 " />
			<path d="M100.0,0.0 L100.0,100.0 " />
			<path d="M100.0,100.0 L0.0,100.0 " />
			<path d="M0.0,100.0 L0.0,0.0 " />

       </g>
    </g>
    
</svg>
```

The counterpart with OpenSCAD would look like:

jmwright, 2023-03-10:
There is [another issue](https://github.com/CadQuery/cadquery/issues/487) talking about a fix for pixel dimensions being forced.

There is also a [draft pull request](https://github.com/CadQuery/cadquery/pull/1277/files) that might fix the dimensions issue.

Paul8043, 2023-03-10:
Thanks for your quick response. Can I try this code in advance or do I have to wait until it is included in 'master'? If possible, I would like to test this code, just to see whether it solves my problem or not. I did a quick inspection of the changed code. The routine "guessUnitOfMeasure(shape)" looks very strange to me. Why can the "UnitOfMeasure" be guessed from the dimensions of the shape? Shapes might have any arbitrary size and the UnitOfMeasure is also an arbitrary definition, in the US probaby "Inch"" and in Europe more likely "Millimeter". But it could also be "Meter", that depends on the application. The plates I process with my CNC-Router might cover an area of 2-3 square meter.

adam-urbanczyk, 2023-03-10:
@Paul8043 use DXF export and SVG for CNC work. SVG export as implemented in CQ is only meant for visualization.

Paul8043, 2023-03-13:
CadQuery offers great features and the code is extremely compact. I like both. Unfortunately my CNC-Router understands only SVG-Files. Your proposal to export the model as DXF and do the conversion to SVG with another tool is a feasible solution, but not my prefered one. I am trying to minimize my tool-chain. CadQuery is the first tool that could achieve this goal, the tool-chain would drop down to a single tool! This would be of great benefit, not only for me, but for the whole cummunity behind my CNC-Router. Again, after inspecting the exporter-code for SVG, I would expect that only small changes are needed to adjust the scaling. I would like to investigate this extension further and hope I will get some help from you. Let me know whether we can  start a closer cooperation.

jmwright, 2023-03-14:
@Paul8043 

> Can I try this code in advance or do I have to wait until it is included in 'master'?

There are several ways to do it. You could check out the PR branch locally ([doc](https://docs.github.com/en/pull-requests/collaborating-with-pull-requests/reviewing-changes-in-pull-requests/checking-out-pull-requests-locally)) then do `pip install -e .` to create a local, editable install. Just be sure to create and activate a virtual environment first. You can also do things with the PYTHONPATH environment variable and a conda environment.

> The routine "guessUnitOfMeasure(shape)" looks very strange to me. 

If I'm not mistaken, that has to do with converting the pixel units to the "user coordinate system". More [here](https://developer.mozilla.org/en-US/docs/Web/SVG/Tutorial/Positions).

adam-urbanczyk, 2023-03-14:
> CadQuery offers great features and the code is extremely compact. I like both. Unfortunately my CNC-Router understands only SVG-Files. Your proposal to export the model as DXF and do the conversion [...]

If you want to contribute, maybe try an SVG version of #1267 ?

Paul8043, 2023-03-14:
Thanks for providing additional information to me. 

@adam-urbanczyk
PR #1267 looks very interesting, because it shows the concept of layers and the possibility to color lines individually. 

@jmwright
Ok, I would like to prepare my notebook for getting the PR-code you mentioned. I will setup a developer environment according "contributing code" as described in your readme. The next step, selecting a specific PR, is not so clear. I know git, but I did not much work with github directly. Can this be done with github-web-page alone or are some low-level git commands necessary. If the later is true, I want first switch the platform from windows to linux. Git is much easier to install under linux. Let me know which platform you recommend.

I have already looked into the svg-document. I think this is the right corner, "width", "height", "viewBox" and "transform" are the items which have to be checked. I will make some experiments with edited svg-files, just to make sure that I have a proper understanding of the scaling.

But getting a proper environment for the PR-code will be the first task.

Paul8043, 2023-03-15:
@jmwright 

> There are several ways to do it. You could check out the PR branch locally [...]

I am trying what you have recommended, but I am faced with a few difficulties. I  work on a notebook with Windows 11 and instead of "conda" I use "mamba". So I am aware that some commands are slightly different.

First of all I want to create a developer environment as described in "Contributing Code". 

Step 1: conda env create -n cq-dev -f environment.yml

It seems that "environment.yml" is an input-file. The --help says "Read package versions from the given file." Unfortunately I don't know much about revisions, at least at this stage in the procedure. Where is "environment.yml" located and what contents should it hold?

So, I executed for

Step 1: mamba create -n cq-dev
Step 2: mamba install -c cadquery -c conda-forge cq-editor=master

Both commands have worked. I have a got an environment with cadquery=master and cq-editor=master.  This should be a solid basis for further work.

Then I have paused the setup procedure, because the next step is totally unclear.

Step 3: pip install -e .

What is the purpose of this command, what does it do and why is it needed. And from which directory has this command to be executed? The pip-doc does not really help. Probably a "cd SomeProject" is needed in front of the pip-command. However at this stage of the procedure it is unclear what is considered as "SomeProject". Up to know I have made some acquaintance with environments and with packages, but not with projects.

adam-urbanczyk, 2023-03-16:
You are trying to develop `cadquery`, so `SomeProject` is supposed to be the folder to which you cloned the repo.

Paul8043, 2023-03-16:
I will spend some time with setting up git and try to understand the used workflow with GitHub. I hope I will be back to CadQuery in a few days.

Thanks so far.