CadQuery/cadquery #1288: Unable to export 3MF files without corruption

RagingRoosevelt, 2023-03-15:
I've tried a couple of approaches, neither of which worked.  I started off trying to export multi-object files.  That didn't work and so too with single-object files.  I'm running CadQuery 2.2.0.

# `Compound`
I've tried the `makeCoumpound` solution discussed in #1165.  This produces a corrupt 3mf file that can't be opened in other programs:
* `Couldn't load that. Error code: 0x80004005` in 3D Builder
* `Cannot insert mesh body.  The source file may be damaged.` in Fusion360

Here's the minimal example:

```python
import cadquery as cq

cq.exporters.export(
    cq.Compound.makeCompound([
        cq.Workplane("front").box(2.0, 2.0, 0.5,centered=[False,False,False]).val(),
        cq.Workplane("back").box(2.0, 2.0, 0.5,centered=[False,False,False]).val()
    ]),
    "3mf_from_compound.3mf"
)
```

I can provide the produced `3mf` file if that's helpful.

# Single-object export

Trying an even simpler example with a single object produces the same errors as above:

```python
cq.exporters.export(
    cq.Workplane("front").box(2.0, 2.0, 0.5,centered=[False,False,False])
    ,
    "3mf_from_assembly.3mf"
)
```

# `Assembly`
I've also tried to use an assembly and that produces the error `AttributeError: 'Assembly' object has no attribute 'tessellate'`.  TBF, I didn't see anything that says that this is supported.

```python
import cadquery as cq

cq.exporters.export(
    cq.Assembly()
    .add(cq.Workplane("front").box(2.0, 2.0, 0.5,centered=[False,False,False]))
    .add(cq.Workplane("back").box(2.0, 2.0, 0.5,centered=[False,False,False]))
    ,
    "3mf_from_assembly.3mf"
)
```

jmwright, 2023-03-15:
Assemblies do not support 3MF export natively at this time, but the way to export an assembly is normally through `cq.Assembly.save`. If you want to export the assembly with `cq.exporters.export()` you need to call `toCompound()` on the assembly first to turn it into a compound. The following export line creates a 3MF that imports fine into PrusaSlicer. `assy` is the variable holding the `cq.Assembly()` object.
```python
cq.exporters.export(assy.toCompound(), "/home/jwright/Downloads/test.3mf")
```

adam-urbanczyk, 2023-03-16:
I tried two random online viewers and your first example rendered correctly. Do you know what the reported errors actually mean?