CadQuery/cadquery #1291: README remove link to out of date vscode extension

lorenzncode, 2023-03-19:
I have not tried it, but according to #1289 the extension mentioned in the README is discontinued.  

The README may not be the best place for keeping up to date with external projects.  I'd suggest to remove links to other projects with an exception for the jupyter-cadquery extension as it has been part of the docs for a longer time.

I would suggest to remove this part entirely.:
```
There are currently 4 different ways to use CadQuery for designing your next project:
* Using the [CQ-editor GUI](https://github.com/CadQuery/CQ-editor)
* From a [Jupyter notebook](https://github.com/bernhard-42/jupyter-cadquery)
* Using a [VSCode extension](https://marketplace.visualstudio.com/items?itemName=roipoussiere.cadquery)
```

There are already existing links to CQ-editor and the jupyter-cadquery.  No need to link again.  Enumerating the different ways to use CadQuery does not appear to be accurate in any case - what about the Docker usage that is mentioned later?

A more appropriate location for other CadQuery projects seems to be the awesome-cadquery repo.

adam-urbanczyk, 2023-03-19:
+1, it'd be better to focus on core CQ things in the README (built-in jupyter lab support and CQ-editor).

jmwright, 2023-03-19:
We also have the [awesome-cadquery](https://github.com/CadQuery/awesome-cadquery) repo for tracking things now, so there is no requirement to have them in the core readme like we did when we were a smaller project and there were not a lot of extensions.

lorenzncode, 2023-05-07:
Documentation updated in #1305.

closed 2023-05-07 by lorenzncode