CadQuery/cadquery #1295: Strange behavior with edge order in face's wire

geiseri, 2023-03-26:
I have a few issues [in this file](https://gitlab.com/geiseri/cadquery_models/-/blob/2f4863124cc42c559a09ead855277930d024d7b1/top_groove.py) but first I need to clarify the order of Edges returned in a wire.  

I am using a face in this test case to grab its outer wire so that I can use it in a future operation to create a sweep.  The sweep has its own issues, but one big one comes from the fact it looks like the order of edges in the wire on line [20](https://gitlab.com/geiseri/cadquery_models/-/blob/2f4863124cc42c559a09ead855277930d024d7b1/top_groove.py#L20) seems to "shuffle" around.  After 4 consecutive rounds running `cq-server` to build an STL I see the following output on line [25](https://gitlab.com/geiseri/cadquery_models/-/blob/2f4863124cc42c559a09ead855277930d024d7b1/top_groove.py#L25).

Run 1

```
edge 0 is [Vertex: (-10.0, -16.75, 2.629867970565564e-16), Vertex: (10.0, -16.75, 2.629867970565564e-16)]
edge 1 is [Vertex: (10.0, -16.75, 2.629867970565564e-16), Vertex: (11.75, -15.000000000000002, 2.629867970565564e-16)]
edge 2 is [Vertex: (11.75, 15.0, 2.629867970565564e-16), Vertex: (11.75, -15.000000000000002, 2.629867970565564e-16)]
edge 3 is [Vertex: (11.75, 15.0, 2.629867970565564e-16), Vertex: (10.0, 16.75, 2.629867970565564e-16)]
edge 4 is [Vertex: (-10.0, 16.75, 2.629867970565564e-16), Vertex: (10.0, 16.75, 2.629867970565564e-16)]
edge 5 is [Vertex: (-10.0, 16.75, 2.629867970565564e-16), Vertex: (-11.75, 15.000000000000002, 2.629867970565564e-16)]
edge 6 is [Vertex: (-11.75, 15.000000000000002, 2.629867970565564e-16), Vertex: (-11.75, -15.000000000000002, 2.629867970565564e-16)]
edge 7 is [Vertex: (-11.75, -15.000000000000002, 2.629867970565564e-16), Vertex: (-10.0, -16.75, 2.629867970565564e-16)]
```

Run 2

```
edge 0 is [Vertex: (11.75, 15.0, 8.76622656855189e-17), Vertex: (11.75, -15.000000000000002, 8.76622656855189e-17)]
edge 1 is [Vertex: (10.0, -16.75, 8.76622656855189e-17), Vertex: (11.75, -15.000000000000002, 8.76622656855189e-17)]
edge 2 is [Vertex: (-10.0, -16.75, 8.76622656855189e-17), Vertex: (10.0, -16.75, 8.76622656855189e-17)]
edge 3 is [Vertex: (-11.75, -15.000000000000002, 8.76622656855189e-17), Vertex: (-10.0, -16.75, 8.76622656855189e-17)]
edge 4 is [Vertex: (-11.75, 15.0, 8.76622656855189e-17), Vertex: (-11.75, -15.000000000000002, 8.76622656855189e-17)]
edge 5 is [Vertex: (-10.0, 16.75, 8.76622656855189e-17), Vertex: (-11.75, 15.0, 8.76622656855189e-17)]
edge 6 is [Vertex: (-10.0, 16.75, 8.76622656855189e-17), Vertex: (10.0, 16.75, 8.76622656855189e-17)]
edge 7 is [Vertex: (11.75, 15.0, 8.76622656855189e-17), Vertex: (10.0, 16.75, 8.76622656855189e-17)]
```

Run 3

```
edge 0 is [Vertex: (-10.0, 16.75, -3.5064906274207524e-16), Vertex: (10.0, 16.75, -3.5064906274207524e-16)]
edge 1 is [Vertex: (11.75, 15.000000000000002, -3.5064906274207524e-16), Vertex: (10.0, 16.75, -3.5064906274207524e-16)]
edge 2 is [Vertex: (11.75, 15.000000000000002, -3.5064906274207524e-16), Vertex: (11.75, -15.000000000000002, -3.5064906274207524e-16)]
edge 3 is [Vertex: (10.0, -16.75, -3.5064906274207524e-16), Vertex: (11.75, -15.000000000000002, -3.5064906274207524e-16)]
edge 4 is [Vertex: (-10.0, -16.75, -3.5064906274207524e-16), Vertex: (10.0, -16.75, -3.5064906274207524e-16)]
edge 5 is [Vertex: (-11.75, -15.000000000000002, -3.5064906274207524e-16), Vertex: (-10.0, -16.75, -3.5064906274207524e-16)]
edge 6 is [Vertex: (-11.75, 15.0, -3.5064906274207524e-16), Vertex: (-11.75, -15.000000000000002, -3.5064906274207524e-16)]
edge 7 is [Vertex: (-10.0, 16.75, -3.5064906274207524e-16), Vertex: (-11.75, 15.0, -3.5064906274207524e-16)]
```

Run 4

```
edge 0 is [Vertex: (11.75, 15.0, 7.012981254841505e-16), Vertex: (11.75, -15.000000000000002, 7.012981254841505e-16)]
edge 1 is [Vertex: (10.0, -16.75, 7.012981254841505e-16), Vertex: (11.75, -15.000000000000002, 7.012981254841505e-16)]
edge 2 is [Vertex: (-10.0, -16.75, 7.012981254841505e-16), Vertex: (10.0, -16.75, 7.012981254841505e-16)]
edge 3 is [Vertex: (-11.75, -15.000000000000002, 7.012981254841505e-16), Vertex: (-10.0, -16.75, 7.012981254841505e-16)]
edge 4 is [Vertex: (-11.75, 15.0, 7.012981254841505e-16), Vertex: (-11.75, -15.000000000000002, 7.012981254841505e-16)]
edge 5 is [Vertex: (-10.0, 16.75, 7.012981254841505e-16), Vertex: (-11.75, 15.0, 7.012981254841505e-16)]
edge 6 is [Vertex: (-10.0, 16.75, 7.012981254841505e-16), Vertex: (10.0, 16.75, 7.012981254841505e-16)]
edge 7 is [Vertex: (11.75, 15.0, 7.012981254841505e-16), Vertex: (10.0, 16.75, 7.012981254841505e-16)]
```

It looks like it comes back directly from the order of wires returned by the `outerWire` result on the face.  It looks like this comes directly from [here](https://github.com/CadQuery/cadquery/blob/2d355177871c0b1a275640659ba203cf3cdf267c/cadquery/occ_impl/shapes.py#L2297). Is this a bug?  Is there maybe a better way to do this?

adam-urbanczyk, 2023-03-27:
Iterators in OCCT (and thus in cq) are not ordered.

geiseri, 2023-03-27:
Is there a way to provide a "start" and "end" vertex and have it sorted from there?  I am not sure how you would sort them though.

adam-urbanczyk, 2023-04-09:
CQ does not provide such functionality. You could sort them yourself, but I don't see why you'd need to specify order when sweeping.