CadQuery/cadquery #1301: OCP with VTK engine

ragdenesie, 2023-03-29:
Hello,
I'm quite a freshman to this project.
For already some while I'm developing a visualization tool around a FE package. The complete package should include result visualization as well as geometry modelling and mesh generation. Initially, the geometry part was implemented on the basis of some internal geometry package. The overall tool is completely pythonic. Already some time ago, I started to work with pyOCCT to enhance the geometry part, the visualization part is based on VTK. For several reasons, it is not possible for me to work with conda, but with virtualenv together with some packages installed in the system or in my home environment. 
However, the pyOCCT is limited to VTK '8.2.0', also I had to do may own wrapping due to version mismatches, and I would like to use the latest'9.x.x' releases of VTK. Previously, I tried OCP from source. The clang part works in the meanwhile but the compilation appears to have some issues (eventually to be discussed separately).
Recently, I figured that OCP and cadquery are available as virtualenv packages. Trying my test routine I found some issue with the VTK interaction, namely IVtkOCC. Somehow, the getVtkPolyData always returns None, whatever done in the code or in the environment settings.
Today, I came across a discussion related to VTK of [adam-urbanczyk], Accordingly, the simplest code would become:
import OCP
from cadquery import Solid
from cadquery import Shape
import OCP
solid_box = Solid.makeBox(10,10,10)
solid=solid_box.wrapped
w=solid
vs = OCP.IVtkOCC.IVtkOCC_Shape(w)
sd = OCP.IVtkVTK.IVtkVTK_ShapeData()
sm = OCP.IVtkOCC.IVtkOCC_ShapeMesher()
sm.Build(vs,sd)
sd.getVtkPolyData()
res=sd.getVtkPolyData()
print(res)
As already mentioned, the res is 'None', while a PolyData would be expected. I have no idea what side effects or additional dependencies could be there. Effectively, a similar test with my homemade pyOCCT, though vtk-8.2.0, always returns a PolyData that may be piped into a InformationObject.
I would be grateful, if anyone could assist in solving this issue.
May be some bug in the virualenv distribution? No idea.
Thanks in advance for your assistance

adam-urbanczyk, 2023-03-29:
I cannot reproduce:

```
print(res)
vtkPolyData (000001CE37AF3E50)
  Debug: Off
  Modified Time: 154
  Reference Count: 2
  Registered Events: (none)
  Information: 000001CE37A42690
  Data Released: False
  Global Release Data: Off
  UpdateTime: 0
  Field Data:
    Debug: Off
    Modified Time: 79
    Reference Count: 1
    Registered Events: (none)
    Number Of Arrays: 0
    Number Of Components: 0
    Number Of Tuples: 0
  Number Of Points: 128
  Number Of Cells: 68
  Cell Data:
    Debug: Off
    Modified Time: 154
    Reference Count: 1
    Registered Events: 
      Registered Observers:
        vtkObserver (000001CE374F5EC0)
          Event: 33
          EventName: ModifiedEvent
          Command: 000001CE37A3D8F0
          Priority: 0
          Tag: 1
    Number Of Arrays: 2
    Array 0 name = SUBSHAPE_IDS
    Array 1 name = MESH_TYPES
    Number Of Components: 2
    Number Of Tuples: 68
    Copy Tuple Flags: ( 1 1 1 1 1 0 1 1 1 1 1 )
    Interpolate Flags: ( 1 1 1 1 1 0 0 1 1 1 1 )
    Pass Through Flags: ( 1 1 1 1 1 1 1 1 1 1 1 )
    Scalars: (none)
    Vectors: (none)
    Normals: (none)
    TCoords: (none)
    Tensors: (none)
    GlobalIds: (none)
    PedigreeIds: (none)
    EdgeFlag: (none)
    Tangents: (none)
    RationalWeights: (none)
    HigherOrderDegrees: (none)
  Point Data:
    Debug: Off
    Modified Time: 144
    Reference Count: 1
    Registered Events: 
      Registered Observers:
        vtkObserver (000001CE374F5E00)
          Event: 33
          EventName: ModifiedEvent
          Command: 000001CE37A3D8F0
          Priority: 0
          Tag: 1
    Number Of Arrays: 1
    Array 0 name = Normals
    Number Of Components: 3
    Number Of Tuples: 128
    Copy Tuple Flags: ( 1 1 1 1 1 0 1 1 1 1 1 )
    Interpolate Flags: ( 1 1 1 1 1 0 0 1 1 1 1 )
    Pass Through Flags: ( 1 1 1 1 1 1 1 1 1 1 1 )
    Scalars: (none)
    Vectors: (none)
    Normals: 
      Debug: Off
      Modified Time: 141
      Reference Count: 2
      Registered Events: (none)
      Name: Normals
      Data type: float
      Size: 765
      MaxId: 383
      NumberOfComponents: 3
      Information: 0000000000000000
      Name: Normals
      Number Of Components: 3
      Number Of Tuples: 128
      Size: 765
      MaxId: 383
      LookupTable: (none)
    TCoords: (none)
    Tensors: (none)
    GlobalIds: (none)
    PedigreeIds: (none)
    EdgeFlag: (none)
    Tangents: (none)
    RationalWeights: (none)
    HigherOrderDegrees: (none)
  Bounds: 
    Xmin,Xmax: (0, 10)
    Ymin,Ymax: (0, 10)
    Zmin,Zmax: (0, 10)
  Compute Time: 206
  Editable: false
  Number Of Points: 128
  Point Coordinates: 000001CE373F65C0
  PointLocator: 0000000000000000
  CellLocator: 0000000000000000
  Number Of Vertices: 8
  Number Of Lines: 48
  Number Of Polygons: 12
  Number Of Triangle Strips: 0
  Number Of Pieces: 1
  Piece: -1
  Ghost Level: 0
  CellsBounds: 
    Xmin,Xmax: (0, 10)
    Ymin,Ymax: (0, 10)
    Zmin,Zmax: (0, 10)
  CellsBounds Time: 207
```

ragdenesie, 2023-03-30:
Hi Adam,
Many thanks for the fast response. Not really unexpected that you may not easily reproduce, because it worked already according to thread #281.
However, I would be grateful in any assistance in solving my issue.
Are you working with the venv distribution as of "https://pypi.org/project/cadquery-ocp/"? I'm suspicious that something happens during the packaging. I send also my version strings:
>>> import OCP
>>> import vtk
>>> import cadquery
>>> OCP.__version__
'7.7.0.0'
>>> vtk.VTK_VERSION
'9.0.1'
>>> cadquery.__version__
'2.2.1.dev21+g2d35517'
>>> 
What other side-effects may be in my environment? Any idea how to figure?
Best wishes
Edgar

adam-urbanczyk, 2023-03-30:
> Hi Adam, Many thanks for the fast response. Not really unexpected that you may not easily reproduce, because it worked already according to thread #281. However, I would be grateful in any [...]
 
> What other side-effects may be in my environment? Any idea how to figure? Best wishes Edgar

I don't work with cadquery-ocp to avoid these kind of issue. https://github.com/CadQuery/ocp-build-system is a better place for this question.

> Not sure whether this could be a hint: Inside Anaconda I find, recently upgraded though:
> 
> > /home/edgar/Anaconda/anaconda3/envs/binder/x86_64-conda-linux-gnu/sysroot/lib64/libc-2.12.so
> 
> Inside the system is to find
> 
> > /usr/lib/i386-linux-gnu/libc-2.31.so
> 
> Hence, I'm suspicous whether the conda environment uses some older libc compared to the system and the virtualenv, although I should upgrade the system during the next weeks.

How is this related your question about cadquery-ocp? Use `ldd` if you need to check which libc is used. Sysroot is used for compiling only AFAIK.

ragdenesie, 2023-05-15:
"# 49" should actually be inside "Cadquery/oywrap", ...
Excuse

Jojain, 2023-05-15:
There has been some investigation to see how feasible it was to build OCP with pyodide but without success, if you want to share you work feel free to share it in this discord thread : https://discord.com/channels/964330484911972403/1079454322221396008
With your experience we might manage to improve on that path.

adam-urbanczyk, 2023-05-15:
Thanks for the update. Given that OCP compiles and works, it is unlikely that you had to run pywrap at all. I'm closing this issue, if you have something OCP/pywrap specific you want to share with the community, please do it in the relevant repos and/or issues.

ragdenesie, 2023-05-15:
Maybe I was not clear enough. It works after serious and tedious work with pywrap to produce proper OCP. In the moment, I have the feeling that you are not really interested in any enhancement of the cadquery environment which of course includes the pywrap to produce a proper working overall environment, actually the basis being OCP. 
Best regards and many thanks for the OCP base work.

⁣BlueMail for Android herunterladen ​

adam-urbanczyk, 2023-05-16:
Please share the details in the relevant repos/issues. You claim that modification of the sources was needed, but they are compiled without any modification in the CI and used all over the place, so some skepticism is justified.

ragdenesie, 2023-05-17:
This is followed up @ thread "#49" inside "CadQuery/pywrap".
-> may be closed by the developers; kept it open for the moment for having it visible

adam-urbanczyk, 2023-06-29:
AFAICT there was some misunderstanding  on how to use pywrap to generate OCP. Long story short, don't try it unless you know what you are doing and use the published OCP sources. If you really insist, look at how it is done in the CI of the OCP repo. With proper configuration no manual edits of the resulting sources are needed.

closed 2023-06-29 by adam-urbanczyk