CadQuery/cadquery #1304: Error on import cadquery with new install without explicitly specifying occt version

lorenzncode, 2023-04-02:
Install command:  `mamba install -c conda-forge cadquery`

It looks like this issue is due to new OCCT 7.7.1.

```
$ python
Python 3.11.0 | packaged by conda-forge | (main, Jan 14 2023, 12:27:40) [GCC 11.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import cadquery
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/lorenzn/tools/mambaforge_230402/envs/cqreltestbad/lib/python3.11/site-packages/cadquery/__init__.py", line 10, in <module>
    from .occ_impl.geom import Plane, BoundBox, Vector, Matrix, Location
  File "/home/lorenzn/tools/mambaforge_230402/envs/cqreltestbad/lib/python3.11/site-packages/cadquery/occ_impl/geom.py", line 5, in <module>
    from OCP.gp import (
ImportError: /home/lorenzn/tools/mambaforge_230402/envs/cqreltestbad/lib/python3.11/site-packages/OCP.cpython-311-x86_64-linux-gnu.so: undefined symbol: _ZN20DE_ConfigurationNode10UpdateLoadEv
```

```
$ conda list "cadquery|ocp|occt"
# packages in environment at /home/lorenzn/tools/mambaforge_230402/envs/cqreltestbad:
#
# Name                    Version                   Build  Channel
cadquery                  2.2.0              pyhd8ed1ab_0    conda-forge
occt                      7.7.1                h6cf717c_0    conda-forge
ocp                       7.7.0.0         py311hb87e267_1    conda-forge
```

adam-urbanczyk, 2023-04-05:
master has a pin for occt after merging #1293

Paul8043, 2023-04-06:
Checked today, healthy again under windows.

bragostin, 2023-04-12:
The issue seems to be still present with Linux:
```
conda list "cadquery|ocp|occt"
# packages in environment at /home/bruno/.conda/envs/cadquery:
#
# Name                    Version                   Build  Channel
cadquery                  master                    py3.9    cadquery
occt                      7.7.1                h6cf717c_0    conda-forge
ocp                       7.7.0.0          py39h5d0cb9e_1    conda-forge
```
and I get the `mportError: /home/bruno/.conda/envs/cadquery/lib/python3.9/site-packages/OCP.cpython-39-x86_64-linux-gnu.so: undefined symbol: _ZN20DE_ConfigurationNode10UpdateLoadEv` error
When I try to force-reinstall `cadquery=master` with conda I get
```
CondaHTTPError: HTTP 404 NOT FOUND for url <https://conda.anaconda.org/cadquery/linux-64/cadquery-master-py3.9.tar.bz2>
```

lorenzncode, 2023-04-12:
@bragostin,   There was a problem with the conda upload after the latest commit yesterday.

As a workaround, if you do not require the latest dev build, you can install the release version 2.2.0 (conda or mamba):
`mamba install -c conda-forge cadquery=2.2.0 occt=7.7.0`

Schievel1, 2023-04-24:
This still happens for me even though I started from a fresh conda install and only did
```
conda create --name=dactyl-keyboard python=3.8 -y
 conda activate dactyl-keyboard
 conda install -c conda-forge cadquery=2.2.0 occt=7.7.0
 pip install dataclasses-json numpy scipy solidpython
 conda update --all -y
```

Then I run a python script to generate keyboards (https://github.com/bullwinkle3000/dactyl-keyboards) and get 
```
dactyl-keyboard on  master [✘!] via  dactyl-keyboard ➜ python src/dactyl_manuform.py
NO CONFIGURATION SPECIFIED, USING run_config.json
Found Current Engine in Config = cadquery
Traceback (most recent call last):
  File "src/dactyl_manuform.py", line 2098, in <module>
    make_dactyl()
  File "src/dactyl_manuform.py", line 131, in make_dactyl
    globals().update(importlib.import_module("helpers_cadquery").__dict__)
  File "/home/pascal/miniconda3/envs/dactyl-keyboard/lib/python3.8/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 843, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/home/pascal/dev/dactyl-keyboard/src/helpers_cadquery.py", line 1, in <module>
    import cadquery as cq
  File "/home/pascal/miniconda3/envs/dactyl-keyboard/lib/python3.8/site-packages/cadquery/__init__.py", line 10, in <module>
    from .occ_impl.geom import Plane, BoundBox, Vector, Matrix, Location
  File "/home/pascal/miniconda3/envs/dactyl-keyboard/lib/python3.8/site-packages/cadquery/occ_impl/geom.py", line 5, in <module>
    from OCP.gp import (
ImportError: /home/pascal/miniconda3/envs/dactyl-keyboard/lib/python3.8/site-packages/OCP.cpython-38-x86_64-linux-gnu.so: undefined symbol: _ZN20DE_ConfigurationNode10UpdateLoadEv
(dactyl-keyboard)
```

lorenzncode, 2023-05-07:
The installation instructions for the latest release have also been updated in #1305.

closed 2023-05-07 by lorenzncode