CadQuery/cadquery #1307: Assembly randomness when using 'Axis' type and an rotational angle

marpoleSim, 2023-04-06:
Below is a script I used in cd-editor. I used Axis type for the second  constraint.

```
height = 60.0
width = 80.0
thickness = 10.0

box1 = cq.Workplane("XY").box(height, width, thickness)
box2 = cq.Workplane("XY").box(height, width, thickness)

# add a direction indicator
box2 = box2.faces(">Y").workplane().circle(1.0).extrude(10)

assy = (
    cq.Assembly()
    .add(box1, name="box1")
    .add(box2, name="box2", color=cq.Color('red'))
)

(
    assy
    .constrain("box1@faces@<Z", "box2@faces@>Z", "Plane")
    .constrain("box1@faces@<Y", "box2@faces@<Y", "Axis", param = 15)
)

assy.solve()
show_object(assy)
```

When I use an angle of 15 degrees, the result looks like below, which is good. 

[image]

But when I change the angle to 15.1 degrees, the result looks like below, where the angle is actually -15.1 degrees. 

[image]

I made a few more tests. It seems the occurrence is random. Though the absolute angle is correct, the result doesn't follow the formula in the cadquery document, where direction is considered. Is this a bug?

[image]

adam-urbanczyk, 2023-04-07:
No bug here, the equation(s) need to be updated. For non zero parameters you'll get a continuum (AFAICT cone) of solutions. If you add a plane you'll get two possible outcomes.

marpoleSim, 2023-04-07:
Thanks. What would be a way to rotate a box relative to another box with only one solution (for example clockwise)?

marpoleSim, 2023-04-08:
I figured it out. This can be made by first creating a dummy face out of the box, rotating the face to the specified angle and then mating the other part to the dummy face.

closed 2023-04-08 by marpoleSim

adam-urbanczyk, 2023-04-09:
Great that you were able to figure this out! Note the following trick:
```python
res = cq.Workplane().box(1,1,1).faces('>Z').rotate((0,0,0),(0,0,1),15).tag('dummy').end(2)
assy.add(res, name='box1')
...
```
will allow you to refer to the dummy face when specifying constraints with the `dummy` tag like so:
```python
assy.constrain("box1?dummy","box2@faces>Z","Axis")
```

If all fails you can always use the following overload and use *any* shapes, not even ones that actually belong to your model
https://github.com/CadQuery/cadquery/blob/58a9c72a6798d9dcc438f391ee9c64e19d135421/cadquery/assembly.py#L290

marpoleSim, 2023-04-10:
Thanks Adam. A much nicer and compact script! Works on my laptop.