CadQuery/cadquery #1308: deprecate_kwarg_name is not type aware

thelittlebug, 2023-04-09:
thats the reason why you get no autocompletion for simple things like the following
```
Workplane().text("You rock", 12, 5,). #at this point the editor does not know how to autocomplete
```

but there is a simple solution to this issue: 
```
TCallable = TypeVar("TCallable", bound=Callable)

class deprecate_kwarg_name:
    def __init__(self, name, new_name):
        self.name = name
        self.new_name = new_name

    def __call__(self, f: TCallable) -> TCallable:
        @wraps(f)
        def wrapped(*args, **kwargs):

            if self.name in kwargs:
                warn(
                    f"Kwarg <{self.name}> will be removed. Please use <{self.new_name}>",
                    FutureWarning,
                )

            return f(*args, **kwargs)

        return cast(TCallable, wrapped)
```

adam-urbanczyk, 2023-04-09:
Could you open a PR?

thelittlebug, 2023-04-10:
yes, i will definitely do this. sry, i hadn't the possibilities  for that while i was posting this. 1-2 days and i'm in my used to work area ;)

voneiden, 2023-06-14:
@thelittlebug I've opened PR #1348 on your behalf, you OK with that?

PR #1348 (2023-07-01): Add TypeVar to deprecate_kwarg to preserve type information
Closes #1308

Committed on behalf of the author @thelittlebug
closed 2023-07-01 by adam-urbanczyk