CadQuery/cadquery #1319: cut not performed

lalebarde, 2023-04-22:
Possibly related to [#260](https://github.com/CadQuery/cadquery/issues/260), the final cut does not happen:

```
import cadquery as cq
from math import pi as pi, sin as sin, cos as cos, tan as tan, atan as atan

r3 = 40
r4 = 80
a = 0
a1 = 28
a2 = 36

result = cq.Workplane("XY" ).circle(109.6).extrude(30)

evidage = result.faces(">Z").workplane() 
evidage = evidage.ellipseArc(x_radius = r3, y_radius = r3, angle1 = a-a1, angle2 = a+a1 , sense = -1, startAtCurrent = False)
evidage = evidage.lineTo(r4 * cos((a-a2)*pi/180), r4 * sin((a-a2)*pi/180))
evidage = evidage.center(0, 0)
evidage = evidage.ellipseArc(x_radius = r4, y_radius = r4, angle1 = a-a2, angle2 = a + a2, sense = 1, startAtCurrent = False)
evidage = evidage.close()
evidage = evidage.extrude(-30-2)

result = result.cut(evidage)
```
I have extruded the solid to be cut 2mm more to make it appear:

[image]

If I change the last line to `result = result.cut(evidage, clean = False)`, nothing is changed.

lalebarde, 2023-04-22:
Even with a simple rect, it doesn't work. So I have doubts if I use it the right way:

```
import cadquery as cq
result = cq.Workplane("XY" ).circle(109.6).extrude(30)
evidage = result.faces(">Z").workplane().rect(50,50).extrude(-30-1)
result = result.cut(evidage)
```

lalebarde, 2023-04-22:
If I take the documentation example [here](https://cadquery.readthedocs.io/en/latest/examples.html), the cut is not performed:

[image]

lalebarde, 2023-04-22:
Strange, I have rebooted and when starting cq-editor, I have the following error:
```
$ cq-editor 
0.07s - Debugger warning: It seems that frozen modules are being used, which may
0.00s - make the debugger miss breakpoints. Please pass -Xfrozen_modules=off
0.00s - to python to disable frozen modules.
0.00s - Note: Debugging will proceed. Set PYDEVD_DISABLE_FILE_VALIDATION=1 to disable this validation.
ICE default IO error handler doing an exit(), pid = 5646, errno = 32

$ cq-editor -Xfrozen_modules=off
usage: cq-editor [-h] [filename]
cq-editor: error: unrecognized arguments: -Xfrozen_modules=off
```

May it be related? How to make it work again?

Jojain, 2023-04-22:
It is performed cq-editor just displays everything. Try to call `show_object` on the item you want to see and it will only display that

lalebarde, 2023-04-22:
ok, it works with the documentation example.

On my code, it does not on my original code:

```
import cadquery as cq
from math import pi as pi, sin as sin, cos as cos, tan as tan, atan as atan

r3 = 40
r4 = 80
a = 0
a1 = 28
a2 = 36

result = cq.Workplane("XY" ).circle(109.6).extrude(30)

evidage = result.faces(">Z").workplane() 
evidage = evidage.ellipseArc(x_radius = r3, y_radius = r3, angle1 = a-a1, angle2 = a+a1 , sense = -1, startAtCurrent = False)
evidage = evidage.lineTo(r4 * cos((a-a2)*pi/180), r4 * sin((a-a2)*pi/180))
evidage = evidage.center(0, 0)
evidage = evidage.ellipseArc(x_radius = r4, y_radius = r4, angle1 = a-a2, angle2 = a + a2, sense = 1, startAtCurrent = False)
evidage = evidage.close()
evidage = evidage.extrude(-30)

result = result.cut(evidage)
#show_object(evidage)
show_object(result)
```

I think this comes from _evidage = result....._

If I change it to the following, it works:

```
evidage = cq.Workplane("XY" ).ellipseArc(x_radius = r3, y_radius = r3, angle1 = a-a1, angle2 = a+a1 , sense = -1, startAtCurrent = False)
```

[image]

Here _evidage_ is built independently of _result_.  But this is not convenient.

adam-urbanczyk, 2023-04-22:
Use `combine=False` or simply `cutBlind`.

```python
import cadquery as cq
from math import pi as pi, sin as sin, cos as cos, tan as tan, atan as atan

r3 = 40
r4 = 80
a = 0
a1 = 28
a2 = 36

result = cq.Workplane("XY" ).circle(109.6).extrude(30)

evidage = result.faces(">Z").workplane() 
evidage = evidage.ellipseArc(x_radius = r3, y_radius = r3, angle1 = a-a1, angle2 = a+a1 , sense = -1, startAtCurrent = False)

evidage = evidage.lineTo(r4 * cos((a-a2)*pi/180), r4 * sin((a-a2)*pi/180))
evidage = evidage.center(0, 0)
evidage = evidage.ellipseArc(x_radius = r4, y_radius = r4, angle1 = a-a2, angle2 = a + a2, sense = 1, startAtCurrent = False)
evidage = evidage.close()
evidage = evidage.cutBlind(-30)
```

lalebarde, 2023-04-23:
Thanks a lot both of you. combine = False is the right solution for me since I have to add complex geometry to the cutting and I find it easier to build the solid to be cut.

```
evidage = evidage.extrude(30, combine = False)
```

closed 2023-04-23 by lalebarde