CadQuery/cadquery #1322: exportSVG produces svgs with sizes that don't match the physical dimensions of the wires.

giannissc, 2023-04-25:
The exportSVG function produced incorectly scaled svgs. The UnitScale has been set to 3.7795 as per the svg spec and creates svgs with the correct size.

For more information see: #1317

giannissc, 2023-04-25:
Issue created @voneiden

adam-urbanczyk, 2023-04-25:
Currently SVG in CQ is meant for visualization only. Use DXF for fabrication.

giannissc, 2023-04-26:
>Currently SVG in CQ is meant for visualization only. Use DXF for fabrication.
So does that mean that the SVG should be the wrong size? In the machine I have access to svg work better than DXF. 

I will be creating a PR for this soon.

adam-urbanczyk, 2023-04-26:
Probably completely new implementation will be needed for your use case. Take a look at the DXF exporter for inspiration and maybe start with implementing a new svg export func/don't touch the old func.

voneiden, 2023-04-26:
I think having an option to output both fixed size (like now) where the model is scaled to fit the given area or output with given aspect ratio would be nice. 

Something to note also is stroke width, ie. if we have a very large model that is rendered at say 1:1  and then scaled down to fit on a web page by a web browser, the stroke width might end up being too thin. Right?