CadQuery/cadquery #1326: split plane center not as expected

lalebarde, 2023-04-25:
In the following example, after the split, I expect the workplane to be in the plane of the split, but it looks like it is centered on the center of gravity of the remaining part, even if I call a `workplane(centerOption="ProjectedOrigin")`:

```
import cadquery as cq
from math import pi as pi, sin as sin, cos as cos, tan as tan, atan as atan
col_limon = (230, 153, 0, 0.5)

result0 = cq.Workplane("XY" ).box(100, 40, 20)
result = result0.faces(">Z").workplane().transformed(rotate=(90, 20, 0)).split(keepTop=True).workplane(centerOption="ProjectedOrigin").circle(20).extrude(20, combine = True)
show_object(result0, options=dict(rgba=col_limon))
show_object(result)
```

[image]

lalebarde, 2023-04-25:
Possibly related, if after the split I call `faces("<Z")`, I expect it is relative to the workplane of the splitted part, but it is like "Z" is the world one:

```
result = result0.faces(">Z").workplane().transformed(rotate=(90, 20, -360/3*0)).split(keepTop=True).faces("<Z").workplane().circle(20).extrude(20, combine = True)
```
[image]

lorenzncode, 2023-04-30:
The Workplane origin *is* where you expect it: (0, 0, 10.0).

`split` returned with the top object on the stack.  Then when you create the `circle` it is created '_for each item on the stack_' not the Workplane origin. 

To create the circle at the origin you could use tags and create an empty Workplane:

```py
import cadquery as cq

result0 = cq.Workplane("XY").box(100, 40, 20)

result = (
    result0.faces(">Z")
    .workplane()
    .transformed(rotate=(90, 20, 0))
    .tag("splitwp")
    .split(keepTop=True)
    .workplaneFromTagged("splitwp")
    .circle(20)
    .extrude(20, combine=True)
)
```

or call `newObject` with empty list:

```py
import cadquery as cq

result0 = cq.Workplane("XY").box(100, 40, 20)

result = (
    result0.faces(">Z")
    .workplane()
    .transformed(rotate=(90, 20, 0))
    .split(keepTop=True)
    .newObject([])
    .circle(20)
    .extrude(20, combine=True)
)
```

For debug, you can break the chain after the split and inspect the stack.:

```py
import cadquery as cq

result0 = cq.Workplane("XY").box(100, 40, 20)

result = (
    result0.faces(">Z")
    .workplane()
    .transformed(rotate=(90, 20, 0))
    .split(keepTop=True)
    )

print(result.vals())         # note stack is not empty
print(result.val().Center()) # this is where the circle would be centered if you do not clear the stack

```